arduino/Arduino #5598: Arduino Uno .wav player

derekstates, 2016-11-16:
I'm using: Arduino Uno WaveShare microSD card reader
Here is my code:
include <SD.h>

include <TMRpcm.h>

include <SPI.h>

include <pcmRF.h>

include <pcmConfig.h>

define SD_ChipSelectPin 10

TMRpcm tmrpcm;
void setup() { Serial.begin(9600); 
tmrpcm.speakerPin = 9; 
if (!SD.begin(SD_ChipSelectPin)) { 
Serial.println("SD fail"); 
return;
 } 
Serial.println("SD card is good"); 
tmrpcm.setVolume(7); 
tmrpcm.play("1.wav");
}
The issue:
The .wav file doesn't seem to be playing. I have tested the circuit/speaker with other basic audio tones using only the Arduino Uno, so I know that those are hooked up correctly. Using other example codes, I have been able to read from the microSD card.
The audio file is 8-bit mono.
I have been doing research for a few weeks to try and figure out the issue, but I haven't had any luck. Any ideas?
I have basic programming experience, but this is one of my first Arduino projects, so I'm hoping that there is just something simple that I am missing.

TMRh20, 2016-11-18:
@derekstates In the future, for problems with specific libraries, you should typically ask for support on forums like http://arduino.cc/forum or open an issue with the library developer. In this case, me, at https://github.com/TMRh20/TMRpcm/issues

Check out the [TMRpcm wiki](https://github.com/TMRh20/TMRpcm/wiki) 

I'm betting you have the SD card & module connected to the Arduino power supply? Typically SD draws a lot more power when you are reading from it at full speed. Try using batteries or a 5v power supply that can handle probably at least 200mA.

matthijskooijman, 2016-11-18:
Indeed, as @TMRh20 says, this is not the right place for this kind of questions, so I'm closing this issue. @TMRh20, thanks for your response nonetheless!

closed 2016-11-18 by matthijskooijman