arduino/Arduino #5601: CTRL-A "Select all" doesn't work in Serial Monitor

stevenmblock, 2016-11-16:
Using CTRL-A to "Select all," followed by CTRL-C to copy it to the paste buffer, doesn't work in the Serial Monitor any more (this, on a PC running Windows).  I can still select items using click, shift, the cursor keys and arrows, but I used to be able to copy the paste buffer contents with just two simple keystrokes.  Why is this failing now?

dlabun, 2016-11-26:
Which IDE and Windows version are you using?

stevenmblock, 2016-11-26:
I am using IDE 6.12 on a Microsoft Surface Pro 3 running Windows 10 Pro edition.

 

Thanks!

 

SMB

================================================
  Dr. Steven M. Block
  S.W. Ascherman Professor of Sciences
  Departments of Biology and Applied Physics
  Gilbert Hall, 371 Serra Mall, M/C 5020
  Stanford University, Stanford, CA 94305-5020
  Tel: (650)-724-4046
  E-mail:  <mailto:sblock@stanford.edu> sblock@stanford.edu
     ----
   /      \ _________o   0 0
  ||=|====|==============|<|}
   \      /              0 0
     ----
=================================================

agdl, 2016-11-29:
You can't do it if you have the Autoscroll enabled and data is printed out, but when data is in steady state you can do it. Do you confirm this behaviour?

stevenmblock, 2016-11-29:
Dear Arturo,

 

Thank you so much for looking into this issue on my behalf!  I’m on a trip to Europe right now, and won’t return until Dec. 6.  Unfortunately, this is not something I can check without an Arduino board attached to my laptop, and I don’t have one with me.  I can confirm that when I experienced the problem that I reported, Autoscroll was enabled in the Serial Monitor (as usual).

 

If you would like me to check after I return, I would be happy to do so.  Alternatively, perhaps you can confirm this on your end, meanwhile, with another device?

I appreciate your attention to this matter!  Thank you.

 

Best wishes,

SMB

================================================
  Dr. Steven M. Block
  S.W. Ascherman Professor of Sciences
  Departments of Biology and Applied Physics
  Gilbert Hall, 371 Serra Mall, M/C 5020
  Stanford University, Stanford, CA 94305-5020
  Tel: (650)-724-4046
  E-mail:  <mailto:sblock@stanford.edu> sblock@stanford.edu
     ----
   /      \ _________o   0 0
  ||=|====|==============|<|}
   \      /              0 0
     ----
=================================================

sandeepmistry, 2017-02-16:
@stevenmblock any updates on this?

stevenmblock, 2017-02-16:
Dear Sandeep,

 

Thanks for your inquiry.  I am presently running Arduino IDE version 8.1 (on Windows 10) and CTRL-A/CRTL-C (select/copy-all) followed by CTRL-V (in Excel)  now seems to be working again from the Serial Monitor Window.  So the problem seems to have gone away.

 

I am still baffled about why it did not work, but I suspect it might have had something to do with a blank line that I’d placed at the end of the column of numbers, using Serial.println(); .  Could this blank line have been confusing the buffer-paste command in Serial Monitor in an earlier version? Dunno.

 

SMB 

 

================================================

  Dr. Steven M. Block

  S.W. Ascherman Professor of Sciences

  Departments of Biology and Applied Physics

  Gilbert Hall, 371 Serra Mall, M/C 5020

  Stanford University, Stanford, CA 94305-5020

  Tel: (650)-724-4046

  E-mail: sblock@stanford.edu <mailto:sblock@stanford.edu> 

     ----

   /      \ _________o   0 0

  ||=|====|==============|<|}

   \      /              0 0

     ----

=================================================

closed 2017-02-17 by agdl

sandeepmistry, 2017-02-17:
@stevenmblock thanks for trying again, since you're unable to reproduce the issue, we've closed the issue for now. If you have more detailed steps to reproduce it in the future, we can re-open the issue.

stevenmblock, 2017-02-17:
Thanks again for your attention to this matter!

Prof. Steven M. Block
Depts  Biological Sciences and Applied Physics
Stanford University
Stanford, CA 94305
(650) 724-4046