arduino/Arduino #5602: Editor still does not handle backslashes at the end of lines

moconoz, 2016-11-17:
This error still occurs in version 1.6.12

const byte Font5x7[6][7] = {
   {0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,},   // 0x20, Space
   {0x0e, 0x08, 0x08, 0x08, 0x08, 0x08, 0x0e,},   // 0x5b, [
   {0x10, 0x10, 0x08, 0x04, 0x02, 0x01, 0x01,},   // 0x5c, \
   {0x0e, 0x02, 0x02, 0x02, 0x02, 0x02, 0x0e,},   // 0x5d, ]
   {0x04, 0x0a, 0x11, 0x00, 0x00, 0x00, 0x00,},   // 0x5e, ^
   {0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x1f,},   // 0x5f, _
   };
  the backslash causes the next line to be ignored corrupting the table

moconoz, 2016-11-17:
The backslash in the previous entry was itself deleted when posting.

There was a backslash on the following line in the code snippet above
{0x10, 0x10, 0x08, 0x04, 0x02, 0x01, 0x01,}, // 0x5c, (backslash here)

facchinm, 2016-11-17:
Hi @moconoz , could you post a sketch showing the problem? If you are pasting code snippets, enclose them between ``` so markdown doesn't mangle them

cmaglie, 2016-11-17:
The `\` at the end of the line means that the current line continue to the next line. In other words this:

```
// This is a comment
```

and this:

```
// This is \
a comment
```

are exactly the same thing. This is a C/C++ standard feature.
The line in your code:
```
   {0x10, 0x10, 0x08, 0x04, 0x02, 0x01, 0x01,},   // 0x5c, \
   {0x0e, 0x02, 0x02, 0x02, 0x02, 0x02, 0x0e,},   // 0x5d, ]
```
is interpreted by the compiler as:
```
   {0x10, 0x10, 0x08, 0x04, 0x02, 0x01, 0x01,},   // 0x5c,    {0x0e, 0x02, 0x02, 0x02, 0x02, 0x02, 0x0e,},   // 0x5d, ]
```
so basically you lost a line because it becames part of the comment.

To workaround your problem you can try to put a trailing space after the `\` (not very good since the space is invisible), or better slightly change your comment style for example:

`// 0x5c, '\'`

or

`// \, 0x5c`

closed 2016-11-17 by cmaglie

moconoz, 2016-11-17:
Martino,

 

It has been resolved through another response. It is a standard feature of C/C++ where the backslash is used as a line continuation. I initially thought that a double backslash would work (ie override the escape character) but it doesn’t

 

The solution is not put the backslash at the end of the line.

 

Cheers

MOC

moconoz, 2016-11-17:
Thanks,

 

I should have known that but it’s been a while since I’ve used that feature.

 

Cheers

MOC