arduino/Arduino #5603: Backspace characters are not rendered in the console

eric-wieser, 2016-11-17:
When working with the chip-kit extension for arduino, the `pic32prog` program is called to program the PIC. The code for this can be seen [here](https://github.com/sergev/pic32prog/blob/master/pic32prog.c#L522) to emit backspace characters to the console, to emulate a progress bar.

Unfortunately, the arduino console shows these characters as backspace symbols, rather than moving the document insertion point back to the start of the progress bar.

If this were changed in core, it would probably need to be [here](https://github.com/arduino/Arduino/blob/master/app/src/processing/app/EditorConsole.java#L131-L135)

per1234, 2020-10-19:
>Implement handling of \b and \r, and hope nothing else is really needed?

`\r` support was added by https://github.com/arduino/Arduino/pull/9954, but `\b` is still not supported.

matthijskooijman, 2020-10-19:
The link in the original report now points to the wrong line, updated links are https://github.com/sergev/pic32prog/blob/41c472fe4f4e3c7a003fb17abfb3e8cda0fbcec2/pic32prog.c#L645-L646 and https://github.com/sergev/pic32prog/blob/41c472fe4f4e3c7a003fb17abfb3e8cda0fbcec2/pic32prog.c#L347

Note that the program will print a line of `.`, then use `\b` to move the cursor back to the first dot, and then overwrite each dot with `#` while the upload progresses. I'm actually a bit surprised about this, I would have expected a backspace character to actually erase the previous character, rather than just moving the cursor back, but maybe this is not how regular terminals work...