arduino/Arduino #5604: Bootloader cannot be built with up-to-date compiler

LukasJaeger307, 2016-11-17:
Dear Arduino-Team,
I am currently trying to build an Arduino-Bootloader with some crypto extensions (secure IoT and so on) and as a base I wanted to use the existing optiboot-bootloader. I am running an Arduino Uno with an Atmega 328P. This monday I downloaded the code and I spent until just now to build it. The reasons were:

- Makefile included optimization flag -mshort-calls, which is no longer supported by avr-gcc

- The hexfile the compiler spit out was too large to flash it into the Atmega's BLS. This was not such a big problem since I need more space for the extended bootloader anyway but if you just use the makefile, this could be quite frustrating.

- When I flashed the hexfile to the MCU, it produced an unexplicably large amount of frame errors when programming a sketch via UART. I could not find the reason for that.

My solution to all those hassles was to roll back the code and the toolchain to a version uploaded to github in 2011. Looks like the hexfiles in the repository weren't touched for a long time so I'd thought I give you a heads-up, that the bootloader has a problem with current compilers. If you wanna fix this, I'd be happy to support you.

NicoHood, 2016-11-17:
You should use the upstream optiboot or lufa booloader. They are way better and improved nowadays anyways. But you are basically right.

cmaglie, 2016-11-18:
The bootloader of the Uno has been out for very long time now: it has been tested by **a lot** of people, many of them build their shield/products testing them with *that* bootloader. Changing it now means only looking for troubles and we are not going to touch it, unless there is a **very good** reason to do so.

The provided source code is the one that we used to build it, at the times, and upgrading it doesn't make much sense becuase it will just make it out of sync with the hex file.

As @NicoHood said you can use the upstream version.

closed 2016-11-18 by cmaglie

WestfW, 2016-12-15:
The "upstream version" is presumably this one: https://github.com/Optiboot
(This does explicitly have the fixes need for new compilers up through at least 4.9.2)
(although several new "board" packages have their own slightly-modified versions.)