arduino/Arduino #5609: Error compiling for board Arduino/Genuino Mega or Mega 2560

jonawald, 2016-11-19:
I just did an up date to Arduino and got this message. I have used this computer and this arduino a few times before without this issue. 

Arduino: 1.6.8 (Mac OS X), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

failed MSpanList_Insert 0x31f020 0x34bf833a5ffa 0x0
fatal error: MSpanList_Insert

runtime stack:
runtime.MSpanList_Insert(0x2c6d60, 0x31f020)
	/usr/local/go/src/runtime/mheap.c:692 +0x8f
runtime.MHeap_Alloc(0x2c6c20, 0x1, 0x1000000000a, 0x10359)
	/usr/local/go/src/runtime/mheap.c:240 +0x66
runtime.MCentral_CacheSpan(0x2ce098, 0x2d21b8)
	/usr/local/go/src/runtime/mcentral.c:85 +0x167
runtime.MCache_Refill(0x307000, 0xc20000000a, 0xc208012000)
	/usr/local/go/src/runtime/mcache.c:90 +0xa0

goroutine 1 [running]:
runtime.switchtoM()
	/usr/local/go/src/runtime/asm_amd64.s:198 fp=0xc2083f3458 sp=0xc2083f3450
runtime.mallocgc(0x90, 0x1bbce0, 0xc200000001, 0x5f)
	/usr/local/go/src/runtime/malloc.go:178 +0x849 fp=0xc2083f3508 sp=0xc2083f3458
runtime.newobject(0x1bbce0, 0xc2083f3670)
	/usr/local/go/src/runtime/malloc.go:353 +0x49 fp=0xc2083f3530 sp=0xc2083f3508
os.Stat(0xc208498bd0, 0x66, 0x0, 0x0, 0x0, 0x0)
	/usr/local/go/src/os/file_unix.go:133 +0x4e fp=0xc2083f3570 sp=0xc2083f3530
arduino.cc/builder/builder_utils.ArchiveCompiledFiles(0xc208038240, 0x5f, 0x1cf630, 0x6, 0xc2084f5e00, 0x19, 0x20, 0xc20811a7e0, 0xc20811a700, 0x318f90, ...)
	/Users/jenkins/jenkins/workspace/arduino-builder-macosx/src/arduino.cc/builder/builder_utils/utils.go:260 +0x141 fp=0xc2083f3698 sp=0xc2083f3570
arduino.cc/builder/phases.compileCore(0xc208038240, 0x5f, 0xc20811a7e0, 0x0, 0x7fff5fbffd00, 0x4, 0x318f90, 0x2c0908, 0x0, 0x0, ...)
	/Users/jenkins/jenkins/workspace/arduino-builder-macosx/src/arduino.cc/builder/phases/core_builder.go:91 +0x79f fp=0xc2083f3818 sp=0xc2083f3698
arduino.cc/builder/phases.(*CoreBuilder).Run(0x2c0908, 0xc20803c510, 0x0, 0x0)
	/Users/jenkins/jenkins/workspace/arduino-builder-macosx/src/arduino.cc/builder/phases/core_builder.go:54 +0x410 fp=0xc2083f3978 sp=0xc2083f3818
arduino.cc/builder.runCommands(0xc20803c510, 0xc2083f3b90, 0x1d, 0x1d, 0x19501, 0x0, 0x0)
	/Users/jenkins/jenkins/workspace/arduino-builder-macosx/src/arduino.cc/builder/builder.go:187 +0x139 fp=0xc2083f3a08 sp=0xc2083f3978
arduino.cc/builder.(*Builder).Run(0xc2083f3d88, 0xc20803c510, 0x0, 0x0)
	/Users/jenkins/jenkins/workspace/arduino-builder-macosx/src/arduino.cc/builder/builder.go:118 +0xef7 fp=0xc2083f3d68 sp=0xc2083f3a08
arduino.cc/builder.RunBuilder(0xc20803c510, 0x0, 0x0)
	/Users/jenkins/jenkins/workspace/arduino-builder-macosx/src/arduino.cc/builder/builder.go:218 +0x49 fp=0xc2083f3d90 sp=0xc2083f3d68
main.main()
	/Users/jenkins/jenkins/workspace/arduino-builder-macosx/main.go:320 +0x1eb2 fp=0xc2083f3f98 sp=0xc2083f3d90
runtime.main()
	/usr/local/go/src/runtime/proc.go:63 +0xf3 fp=0xc2083f3fe0 sp=0xc2083f3f98
runtime.goexit()
	/usr/local/go/src/runtime/asm_amd64.s:2232 +0x1 fp=0xc2083f3fe8 sp=0xc2083f3fe0
arduino-builder returned 2
Error compiling for board Arduino/Genuino Mega or Mega 2560.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

dlabun, 2016-11-20:
It doesn't look like you're using the latest version of Arduino... Are you sure you're on the latest version?

closed 2016-11-21 by jonawald

per1234, 2017-11-14:
Duplicate of https://github.com/arduino/Arduino/issues/5360