arduino/Arduino #5616: Issues Regarding Arduino Uno

Jee77, 2016-11-22:
c:\users\jeeso\downloads\arduino-1.6.12-windows\arduino-1.6.12\hardware\tools\avr\bin\../lib/gcc/avr/4.9.2/../../../../avr/bin/ar.exe: unable to rename 'core\core.a'; reason: Permission denied

every time I try to up load the code onto Arduino uno I keep getting this error message and I don't know whats wrong! can someone help me out?

facchinm, 2016-11-22:
Hi @Jee77 ,
could you post the whole verbose output of the compilation?

andresquintero2017, 2017-02-13:
Arduino:1.8.1 (Windows 10), Tarjeta:"Arduino/Genuino Uno"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\usuario\Documents\Arduino\libraries -fqbn=arduino:avr:uno -ide-version=10801 -build-path C:\Users\usuario\AppData\Local\Temp\arduino_build_859997 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose C:\Users\usuario\Documents\Arduino\led8_var_int\led8_var_int.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\usuario\Documents\Arduino\libraries -fqbn=arduino:avr:uno -ide-version=10801 -build-path C:\Users\usuario\AppData\Local\Temp\arduino_build_859997 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose C:\Users\usuario\Documents\Arduino\led8_var_int\led8_var_int.ino
Using board 'uno' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Using core 'arduino' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Detecting libraries used...
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10801 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\standard" "C:\Users\usuario\AppData\Local\Temp\arduino_build_859997\sketch\led8_var_int.ino.cpp" -o "nul"
Generating function prototypes...
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10801 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\standard" "C:\Users\usuario\AppData\Local\Temp\arduino_build_859997\sketch\led8_var_int.ino.cpp" -o "C:\Users\usuario\AppData\Local\Temp\arduino_build_859997\preproc\ctags_target_for_gcc_minus_e.cpp"
"C:\Program Files (x86)\Arduino\tools-builder\ctags\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\Users\usuario\AppData\Local\Temp\arduino_build_859997\preproc\ctags_target_for_gcc_minus_e.cpp"
Compilando programa...
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10801 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\standard" "C:\Users\usuario\AppData\Local\Temp\arduino_build_859997\sketch\led8_var_int.ino.cpp" -o "C:\Users\usuario\AppData\Local\Temp\arduino_build_859997\sketch\led8_var_int.ino.cpp.o"
Compiling libraries...
Compiling core...
Utilizando archivo previamente compilado: C:\Users\usuario\AppData\Local\Temp\arduino_build_859997\core\wiring_pulse.S.o
Utilizando archivo previamente compilado: C:\Users\usuario\AppData\Local\Temp\arduino_build_859997\core\WInterrupts.c.o
Utilizando archivo previamente compilado: C:\Users\usuario\AppData\Local\Temp\arduino_build_859997\core\hooks.c.o
Utilizando archivo previamente compilado: C:\Users\usuario\AppData\Local\Temp\arduino_build_859997\core\wiring.c.o
Utilizando archivo previamente compilado: C:\Users\usuario\AppData\Local\Temp\arduino_build_859997\core\wiring_analog.c.o
Utilizando archivo previamente compilado: C:\Users\usuario\AppData\Local\Temp\arduino_build_859997\core\wiring_digital.c.o
Utilizando archivo previamente compilado: C:\Users\usuario\AppData\Local\Temp\arduino_build_859997\core\wiring_pulse.c.o
Utilizando archivo previamente compilado: C:\Users\usuario\AppData\Local\Temp\arduino_build_859997\core\wiring_shift.c.o
Utilizando archivo previamente compilado: C:\Users\usuario\AppData\Local\Temp\arduino_build_859997\core\CDC.cpp.o
Utilizando archivo previamente compilado: C:\Users\usuario\AppData\Local\Temp\arduino_build_859997\core\HardwareSerial.cpp.o
Utilizando archivo previamente compilado: C:\Users\usuario\AppData\Local\Temp\arduino_build_859997\core\HardwareSerial0.cpp.o
[... cut ...]
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc-ar" rcs  "C:\Users\usuario\AppData\Local\Temp\arduino_build_859997\core\core.a" "C:\Users\usuario\AppData\Local\Temp\arduino_build_859997\core\wiring_pulse.c.o"
c:\program files (x86)\arduino\hardware\tools\avr\bin\../lib/gcc/avr/4.9.2/../../../../avr/bin/ar.exe: unable to rename 'C:\Users\usuario\AppData\Local\Temp\arduino_build_859997\core\core.a'; reason: Permission denied

exit status 1
Error compilación en tarjeta Arduino/Genuino Uno.

facchinm, 2017-02-14:
Would be nice to know how @MaxPowe solved this issue. Running as administrator is a way but it shouldn't be necessary. Which AntiVirus are you running?

andresquintero2017, 2017-02-14:
avg antivirus @facchinm

facchinm, 2017-02-14:
Try pausing it when you compile the sketch; if the compilation succeeds, he's the culprit

andresquintero2017, 2017-02-16:
ESO ERA MUCHAS GRACIAS

sandeepmistry, 2017-02-16:
Closing as resolved, the problem was antivirus related.

closed 2017-02-16 by sandeepmistry

jomardyan, 2019-06-27:
Disabling Antivirus not help.

Mayoogh, 2021-04-18:
Temporary fix 
- Create a new blank  sketch 
```
void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```
- Verify/Compile 
- Close new sketch and try compiling project sketch

mirfozil99, 2021-05-26:
hi Guys ,Mayoogh`s way works thanks