arduino/Arduino #5618: git synchronization proposal

MartinHN, 2016-11-22:
would it be useful for anyone else than myself to commit and embed the url and git hash in EEPROM (or code) every time a new upload is done to a piece of hardware.
such method, would allow us to retrieve current code on whatever duino even if it's laying on your shelf for months (or years..).

pro:
a piece of hardware is linked to it's code without having to store it internally
with simple serial protocol, arduino could open the uploaded version if asked

cons : 
not sure it will be beginner friendly (the arduino way...)

if it has to stay outside of core arduino, can arduino app execute some scripts (triggered by hooks or else..) before uploading to implement such functionnality without modifying arduino code?

facchinm, 2016-11-22:
Hi @MartinHN ,
you surely can add this to your custom flow if needed :smile: 

By following https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5-3rd-party-Hardware-specification#pre-and-post-build-hooks-since-ide-165 you can inject the git tag as an header file (or maybe as a define, as you prefer) , while the sketch should take care of allocating it in the required location. I don't believe most people would benefit from such a flow in the IDE itself but it's surely a useful addition with a lot of use cases!

matthijskooijman, 2016-11-22:
I've also been looking at something like this (but thought about adding the git hash in a define to embed it in the code, not EEPROM). I've also thought about using the hooks, but never actually tried to implement it. @MartinHN, if you try this, let us know how it goes (even if you can't make it to work, perhaps the hooks can be improved in some way).

per1234, 2020-10-07:
Although I think it's a very interesting topic, I'm going to close this because I don't think there's anything more Arduino can do to enable this beyond the hooks they've provided.

I'll echo matthijskooijman in saying that I would welcome specific feature requests for improvements needed to facilitate this sort of use case.

closed 2020-10-07 by per1234