arduino/Arduino #5626: Serial monitor and plotter

agdl, 2016-11-23:
Now if the serial monitor is open and you want to open the plotter (or viceversa) you can't do it simply clicking on the desired extension, but you have to close the opened one first. Would be very nice to autoclose the open one

cousteaulecommandant, 2016-11-29:
Being able to use both monitor and plotter simultaneously would be too complicated to code, right?  If not, it would be nice.  (I guess you can't just have 2 threads accessing the same serial port; there should be some sort of centralized program that reads the serial port and sends data to both windows for this.)

madsdyd, 2020-04-07:
@per1234 I would think this should be closed as a duplicate of #4090

per1234, 2022-11-18:
Obviated by addition of support for using Serial Monitor and Serial Plotter simultaneously in Arduino IDE 2.x

closed 2022-11-18 by per1234