arduino/Arduino #5629: Code Sharing

jhtrzcinski, 2016-11-23:
.ino files can't be opened/saved easily from emails

matthijskooijman, 2016-11-23:
Can you be a bit more specific? What is the problem? What would you see different?

harshshah903, 2016-11-24:
No i think there no plugins available to view it on browser.if there is avaliable let me know.

jhtrzcinski, 2016-11-24:
I emailed my self a code I was working on so I could continue at home. When I tried to open/save the code from my email, it wouldn't save or open. I think it said it needed a specific file folder for it to be opened from or something along those lines. I was wondering if there was already some solution to this problem or if a solution could be implemented into the next update for the Arduino Code Editors.

harshshah903, 2016-11-24:
That's what i told you mate , you need a browser plugin to do it . or if you are familiar with java or python then make one pluging for all of us .

dlabun, 2016-11-24:
I believe the issue you are running into is that Arduino puts each INO file into its own folder. Next time you try saving a INO file from your email place it in its own folder with the same name as the INO file.

lmihalkovic, 2016-11-24:
I ahad made proposal to have .inz that would be a comple ino forder in a zip file. On mh local version i made all the examples work like that, and it is not technically difficult to modify the codebase to work off zip files when u dont care to make any source changes (and lazily expand when u start editing a poject)

closed 2016-11-25 by jhtrzcinski