arduino/Arduino #5632: [Library Manager] : please add ArduinoLed library

biagiom, 2016-11-24:
Github link of ArduinoLed library : https://github.com/biagiom/ArduinoLed
Thanks in advance,
Biagio.

cmaglie, 2016-11-28:
Hi @biagiom 
your library is named "Led" but we are not accepting libraries with name that are too generic.
May I ask you to change the name to something more specific, something like "BM Led Library"?

biagiom, 2016-11-30:
Hi @cmaglie 
I named my library "ArduinoLed" but If you think that it is a generic name I suggest to change its name to "LedBlinky" since it has been developed to control and blink a led attached to an Arduino board.
What do you think about the new name ?
Best regards,
Biagio.

biagiom, 2016-12-01:
Hi @cmaglie 
I have changed the library name to LedBlinky and I also updated all the files related to the old name.
The new GitHub link of the library is https://github.com/biagiom/LedBlinky.
Best regards,
Biagio.

cmaglie, 2016-12-01:
Hi @biagiom,

great, I'm going to add the library to the indexer.
Please remember to tag the latest version 1.3.1, since I can't see it yet in

https://github.com/biagiom/LedBlinky/releases

thanks!

cmaglie, 2016-12-01:
Ok the library is now added, once you add the tag to release 1.3.1 that version will be published within an hour.

closed 2016-12-01 by cmaglie

biagiom, 2016-12-02:
Just added a tag to the release 1.3.1
Thanks a lot and best regards,
Biagio.