arduino/Arduino #5638: unix time functions out by 30 years

raburton, 2016-11-26:
Conversion between `time_t` and `struct tm` formats is out by 30 years, but is consistent in both directions so unless you transfer `time_t` timestamps off the device you probably won't notice.

  ```
  time_t unixtime = 1480171640; // Sat, 26 Nov 2016 14:47:20 GMT
  struct tm time;
  gmtime_r(&unixtime, &time);
  Serial.print("1480171640 = ");
  Serial.print(time.tm_hour, DEC);
  Serial.print(':');
  Serial.print(time.tm_min, DEC);
  Serial.print(':');
  Serial.print(time.tm_sec, DEC);
  Serial.print(' ');
  Serial.print(time.tm_mday, DEC);
  Serial.print('/');
  Serial.print(time.tm_mon + 1, DEC); // tm uses 0 based months
  Serial.print('/');
  Serial.println(time.tm_year + 1900, DEC); // tm years start from 1900
```

Output: `1480171640 = 14:47:20 26/11/2046`

facchinm, 2016-11-28:
Hi @raburton, which board are you targeting? Since "time.h" functions come with libc we can't do much except reporting a possible bug. We use the libc implementation quite widely but never encountered such a problem

raburton, 2016-11-28:
Uno. The functions seen consistent within the library, so converting
backwards and forwards between struct tm and time_t work OK on the device.
I only noticed the problem once I wrote an android app that passed time_t
to the uno, that's when I discovered that the value of time_t isn't
actually correctly on the device.

facchinm, 2016-11-28:
Found it. For the avr-libc implementation `time_t` is Y2K Epoch, so it's 30 years after UNIX Epoch. Not sure if it's intended or a bug, anyway the reference is quite clear.  http://www.nongnu.org/avr-libc/user-manual/group__avr__time.html#ga3346b04b0420b32ccf6b706551b70762 
I double checked on ARM boards (newlib) and the function behaves correctly indeed.

closed 2016-11-28 by facchinm

raburton, 2016-11-28:
Hmm, a bit non-standard but it does appear to be deliberative decision so thanks for clarifying that. Easy enough to use an different epoch in the app now I know that's how it's supposed to be.