arduino/Arduino #5641: Board  Manager install downloads toolsDependencies even if they are the same version as built in.

sleemanj, 2016-11-27:
In Arduino IDE 1.6.13 we have

    $ cat builtin_tools_versions.txt
    arduino.avrdude=6.3.0-arduino6
    arduino.avr-gcc=4.9.2-atmel3.5.3-arduino2

In my Board Manager JSON I have

    "toolsDependencies": [
      {
        "packager": "arduino",
        "name": "avrdude",
        "version": "6.3.0-arduino6"
      },
      {
        "packager": "arduino",
        "name": "avr-gcc",
        "version": "4.9.2-atmel3.5.3-arduino2"
      }
    ]

I would have expected that installing that package with the board manager would have seen that the depended versions are the same as the built in versions and just used the built in version.

But what actually happens is the board manager goes and downloads those avrdude and gcc versions separately and installs them into the user's preferences folder, again, even though they are already the built in versions.

dlabun, 2016-11-27:
The board manager was changed recently to strictly follow the JSON data. You may  want to look at this pull request: https://github.com/arduino/Arduino/pull/5199

cmaglie, 2016-12-21:
> I would have expected that installing that package with the board manager would have seen that the depended versions are the same as the built in versions and just used the built in version.

If, for any reason, you change the IDE (by upgrading or downgrading), the version of the bundled tools may change leaving the AVR core without the required tools. For this reason the tools are downloaded anyway when you install the AVR core via board manager to guarantee that they are always available, even when you change the IDE (and consequently the bundled tools).

closed 2016-12-21 by cmaglie

sleemanj, 2016-12-21:
Yes certainly it needs to have the required tools available for the core when you change IDE version, but if the IDE already has the tools version required (bundled or no) when you install said core, it should simply copy the one it has into the appropriate place for posterity, instead of downloading it again from source.

cmaglie, 2016-12-21:
I see, BTW this means adding other complexity to the installer: to achieve that we should add an exception to check if the required tool is available as bundled and, in that case, copy the tool from the bundle to the staging folder (also we should copy instead of unpack -> another install method to do/test/maintain).

The current install procedure for the other generic cores, instead, just *happens* to do the correct thing without any special case. IMHO it's better to not add complications just to optimize the (very low) cost of another download.