arduino/Arduino #5645: Regarding port visible

chintu74123, 2016-11-28:
Sir/mam,
I have installed 1.6.11 version arduino in win7 32 bit from official website. While dumping a program port is not visible I had installed all drivers but port is not visible. How can i resolve that issue ?

facchinm, 2016-11-28:
Hi @chintu74123 , could you be more explicit? For example, which board are you using? What do you mean with "dumping a program" ? If you need help, the best place is the Arduino [forum](http://forum.arduino.cc/) . This place is intended for reporting bugreports, so the more explicit they are, more easily we can find and solve them. Thanks!

sandeepmistry, 2017-02-16:
Closing this for now due to lack of activity.

closed 2017-02-16 by sandeepmistry