arduino/Arduino #5647: Is there a BLE DFU tool?

mrquincle, 2016-11-28:
Hi all,

We're interested in contributing to Arduino for a BLE (Bluetooth Low Energy) smart plug and beacon.

1. Is there already a BLE DFU tool?
2. Is there already nRF52 support?
3. Is there already mesh library support?
4. Are there headers (preferred setups) for key provisioning?

If the answer is no to all of them, I think we can contribute it in a valuable manner.

Anne

facchinm, 2016-11-29:
Hi @mrquincle , 
a core targeting nRF52 is already here https://github.com/sandeepmistry/arduino-nRF5 . I'm adding the author ( @sandeepmistry ) to the loop so he can answer the other questions. 
BTW, it would be extremely interesting, so thanks in advance for your contribution!

mrquincle, 2016-11-29:
Ah, cool. I've used https://github.com/sandeepmistry/noble I think two years ago! That might be actually a way towards a cross-platform DFU tool. I've seen https://punchthrough.com/blog/posts/developing-cross-platform-ble-apps-our-experience do the same.

However, there might also be a cross-platform python lib available. Or even, hold your horses, support directly in the browser: https://googlechrome.github.io/samples/web-bluetooth/index.html. Although I'm not so sure how that could be invoked directly from the Arduino IDE.

sandeepmistry, 2017-02-16:
I'm closing this for now, as there are no official Arduino boards that support DFU via BLE. This feature should probably be developed on a per third part core basis.

closed 2017-02-16 by sandeepmistry

mrquincle, 2017-02-16:
I interpret you closing this also in the sense that the official Arduino community is not being interested in this code. If this is incorrect, feel free to correct me.