arduino/Arduino #5649: Add serial plotter button (or option to switch in serial monitor)

cousteaulecommandant, 2016-11-29:
Right now the serial plotter is kind of hidden in the Tools menu.  I suggest adding a Serial Plotter button next to the Serial Monitor button on the top right corner of the IDE.

Alternatively, it might be a better idea to make the serial monitor have an option to switch to serial plotter and back (via a button or two tabs), so that both tools are combined on a single window.  (This would indirectly solve #5626.)

madsdyd, 2020-04-07:
@per1234 I should think this should be closed as a duplicate of 4090

cousteaulecommandant, 2020-04-10:
> @per1234 I should think this should be closed as a duplicate of 4090

I wouldn't say this is remotely related to #4090; that one is about "Serial plotter and serial monitor cannot be used simultaneously" and this one is about "There should be a shortcut in serial monitor to switch to serial plotter and vice versa".
Although I guess "merge serial plotter and serial monitor into a single applet" would address all issues.  Maybe it's a good idea to create that feature request as a separate issue.

per1234, 2020-12-04:
Vaguely related to https://github.com/arduino/Arduino/pull/11028, which added the capability to use a Shift + click on the Serial Monitor button to open Serial Plotter

per1234, 2022-11-18:
Closing as resolved by https://github.com/arduino/arduino-ide/pull/1194

closed 2022-11-18 by per1234