arduino/Arduino #5654: [Suggestion] HardwareSerial and other Libraries to extend the Print abstract class

RedBow, 2016-11-30:
I am writing a heavier logger library with more features than the existing ([Logging library](http://playground.arduino.cc/Code/Logging)). I was trying to pass `Serial` to a function taking a `Print` type, noticed later on in my code that something was wrong.
In the sketch as such
```C++
PrintLogWriter p(Serial)
```
and in the implementation (`this->print` is of type `Print&`)
```C++
PrintLogWriter(Print& p)
{
    this->print = p;
    Serial.println((int)&Serial); //995
    Serial.println((int)&print); //0
}
```
so I tried
```C++
PrintLogWriter(Print* p)
{
    this->print = *p;
    Serial.println((int)&Serial) //995
    Serial.println((int) p); //995
    Serial.println((int)&print) //0
}
```
so it was obvious that something was up with the cast (especially when `(Print)Serial` threw a compile time exception), and when I took a look at the `HardwareSerial.h` file, it extended `Stream`, but not `Print`. So I'm not sure if this is because someone objected to multiple inheritance,  or because of some legitimate design choice, but I think it would be convenient if `Serial` could be casted to `Print`. I also noticed the same of EthernetClient.
For now this is not a giant obstacle, as I can simply make a wrapper for it, however, for future usage, I thought it would be nice.

matthijskooijman, 2016-11-30:
The Stream class already extends the Print class, so what you're trying should work. Can't see what's wrong with your code directly, though. `(Print)Serial` is expected to not work, since you're casting to a different object type, which would create a new `Print` object, which is abstract. You probably intended `(Print&)Serial` or `(Print*)&Serial` there.

I wonder if in ` Serial.println((int)&print); //0`, the `print` variable is perhaps somehow shadowed by another variable, or not being resolved properly. Does it also break if you do things like `this->print.println("Foo");` ?

sleemanj, 2016-12-02:
I think that you would have to use an initialisation list to get what you want rather than do it in the constructor function.

    class PrintLogWriter
    {
        PrintLogWriter(Print& p) : print(p) {}
        Print& print;
    };

A reference is not a pointer, you can't change what the reference is.. err.. referencing (I was going to say pointing to, but that would be even more confusing), this can only be set when the reference is initialised.  

Any other time you do `[thereference] = [something]` you are not reassigning the reference but operating on the data that the reference is already supposedly referencing.

By the time your constructor function itself is called, the raw initialisation of the variables is already done so your time for initialising the reference has probably expired.

matthijskooijman, 2016-12-03:
@sleemanj, good catch, I indeed think that's the problem. @RedBow, I think there is no bug here. If you agree, could you close this issue?

RedBow, 2016-12-04:
Wow. I'm super stupid... I can't believe I didn't catch that.

at the same time tho, I'm a little confused about the decision to have `Stream` inherit from `Print`. My understanding was that not everything that can be read can also be written to, and not everything that can be written to can also be read. (like `IStream` and `OStream` of std::C++)

closed 2016-12-04 by RedBow