arduino/Arduino #5660: MCP3221 Driver: New Library Contribution

nadavmatalon, 2016-12-01:
Hi,

I created (and tested with hardware) a new library for driving the MCP3221 (12-BIT ADC with I2C interface).

The library comes complete with testing & usage sketch examples and functional documentation.

The current version of the library is: v1.0.0 (for which a 'Github Release' has been created)

The Github repo of the library can be found at: https://github.com/nadavmatalon/MCP3221

Would be great if it could be added to the Arduino Playground's 'Interfacing with Hardware' list.

Thanks,
Nadav

per1234, 2016-12-01:
>Would be great if it could be added to the Arduino Playground's 'Interfacing with Hardware' list.

I've added your library to that page of the Playground as requested. Actually the Arduino Playground is a public wiki so anyone can edit it but this issue report is necessary for the library to be added to the Arduino IDE [Library Manager](https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ).

nadavmatalon, 2016-12-01:
Thanks!i 
So for future reference, should I simply add my relevant libraries to the Arduino Playground myself by editing the wiki, but still create an issue here so that they also get included in the library manager list?
Best,
Nadav

cmaglie, 2016-12-01:
The library has been added to the library manager.

> So for future reference, should I simply add my relevant libraries to the Arduino Playground myself by editing the wiki, but still create an issue here so that they also get included in the library manager list?

Yes, correct.

closed 2016-12-01 by cmaglie