arduino/Arduino #5664: [Website] Reference Wire Library Page improvement

SukkoPera, 2016-12-01:
We get a lot of threads on the Italian forum about I2C communication problems, which almost always end up with missing pull-up resistors on the SDA/SCL lines.

We cannot really blame the posters, since there is no mention of such pull-ups on the [reference's page for the Wire library](https://www.arduino.cc/en/Reference/Wire), so I think this should be added to the page. In general, I find this page a bit too concise. Compare it to the [SPI page](https://www.arduino.cc/en/Reference/SPI), for instance.

Recently we have even found out that the Mega has onboard pull-ups on the I2C pins: did anybody know that? It is not mentioned anywhere! Not even on the [product page](https://www.arduino.cc/en/Main/ArduinoBoardMega2560)! This also somehow interferes with using pins 20/21 as "normal" pins, so it should really be made much more clear. Maybe the Mega product page could be improved as well.

akash73, 2017-02-10:
@SukkoPera thanks for notice.
Page updated.

closed 2017-02-10 by akash73

SukkoPera, 2017-02-10:
Actually I can't see any changes in the pages I menthoned,

oqibidipo, 2017-02-10:
I see this on the Wire reference page now:
>  Please note that a pull-up resistor is needed when connecting SDA/SCL pins. Please refer to the examples for more informations. MEGA 2560 board has pull-up resistors on pins 20 - 21 onboard.

SukkoPera, 2017-02-10:
I can see that too, now.