arduino/Arduino #5665: dht22 no longer working with 1.6.13

jjtoard, 2016-12-01:
I compiled one of my projects with 1.6.13 and it doesn't work.
I installed the older version 1.6.9 and I don't have issues.
I use the following libraries but it looks only to impact the DHT22 sensor and not the display, relay, ...
On the DHT22 the error handling stops a "sync timedout". 

#include <Wire.h>
#include <LCD.h>
#include <LiquidCrystal_I2C.h>  // F Malpartida's NewLiquidCrystal library
#include <DHT22.h>

facchinm, 2016-12-02:
Hi @jjtoard ,
DHT library from Adafruit works correctly with latest AVR toolchain. Which library are you using? Could you paste here a sample sketch showing the regression? Thanks!

jjtoard, 2016-12-02:
it just doesn't work when uploading with 1.6.13 version.  Installed 1.6.9 and works fine.
thought it was related to windows 10 I run 1.6.13 on while 1.6.9 was runnig on windows 8
After installin 1.6.9 on windows 10, jut uploaded and worked
[DHT22.zip](https://github.com/arduino/Arduino/files/627933/DHT22.zip)

[Ventilator_sturing.zip](https://github.com/arduino/Arduino/files/627929/Ventilator_sturing.zip)

facchinm, 2016-12-06:
It looks like the culprit is not the new compiler but the LTO optimizations. I'll take a closer look to understand what is happening; it would be the first report of a function, not being completely removed, that exhibit strange behaviours.
In the meantime, @jjtoard , you can modify the file `DHT22.cpp` so the `readData` function becomes 
```C++
DHT22_ERROR_t __attribute__((optimize("O0"))) DHT22::readData()
```

@matthijskooijman @PaulStoffregen 
for reference, the library is https://github.com/nethoncho/Arduino-DHT22 and the default example is enough to trigger the issue; https://github.com/arduino/Arduino/pull/5624/commits/35e45c9fe09279e1d5736032dad2dc892e35b6fe is needed to get debug symbol from the lto elf

facchinm, 2016-12-06:
Another way to solve the issue (and probably the key for the "bug") is this one
```patch
diff --git a/DHT22.cpp b/DHT22.cpp
index f0e36cd..f3e4a82 100644
--- a/DHT22.cpp
+++ b/DHT22.cpp
@@ -161,7 +161,7 @@ DHT22_ERROR_t DHT22::readData()
         return DHT_ERROR_SYNC_TIMEOUT;
       }
       retryCount++;
-      delayMicroseconds(2);
+      delayMicroseconds(4);
     } while(!DIRECT_READ(reg, bitmask));
     // Measure the width of the data pulse
     retryCount = 0;
@@ -172,7 +172,7 @@ DHT22_ERROR_t DHT22::readData()
         return DHT_ERROR_DATA_TIMEOUT;
       }
       retryCount++;
-      delayMicroseconds(2);
+      delayMicroseconds(4);
     } while(DIRECT_READ(reg, bitmask));
     bitTimes[i] = retryCount;
   }
```

it looks like LTO is optimizing the loop so the timings are different; this modification is compatible with older cores so I believe the original timeouts are too strict anyway.

jjtoard, 2016-12-07:
Thanks, this indeed looks to work.
I would never have found this one.  Thanks for helping me.

closed 2016-12-15 by facchinm