arduino/Arduino #5672: Uploading process stuck on Due if sketch is larger then 256KB and 'Verify code after upload' is set

prasimix, 2016-12-03:
An attempt to upload sketch on Due that is larger then 256KB cannot be completed if `Verify code after upload` option is selected:

[image]

To reproduce this step, make sketch that is more then 256KB long or try this [eez_psu_sketch](https://github.com/eez-open/psu-firmware) where we noticed this bug, go to **File... Preferences** and select `Verify code after upload` option and start upload (Ctrl+U).

facchinm, 2016-12-05:
Hi @prasimix ,
thanks for reporting. I tested the sketch (both with and without verify) and it behaves correctly (on Linux). I'm attaching the (super) verbose output in case you are interested or can spot any relevant difference (like the OS in use :slightly_smiling_face: ).
[verbose_out.txt](https://github.com/arduino/Arduino/files/631807/verbose_out.txt)

If you enable the verbose output you should see the `bossac` version; mine is `1.6.1`

prasimix, 2016-12-05:
Thanks for quick response. How can I generate such verbose output file? I tried uploading (not just compiling) once again with selected both *Show verbose output during: upload* and *Verify code after upload* options and got the same problem.

mvladic, 2016-12-07:
I tried with `Arduino IDE 1.6.9` on Windows 10 and upload fails during verify as @prasimix reported. Here is the verbose output:

[verbose_out.zip](https://github.com/arduino/Arduino/files/636711/verbose_out.zip)

You can see that `bossac` is version `1.6.1`.

facchinm, 2016-12-07:
@mvladic @prasimix I may have an idea; how long is it taking the upload process? We recently added a check on the IDE to avoid stuck programmers instances which block the serial port (https://github.com/arduino/Arduino/blob/ec2e9a642a085b32701cf81297ee7c1570177195/arduino-core/src/cc/arduino/packages/Uploader.java#L136)
If your upload takes more than 2 minutes it gets killed forcibly... On my PC it takes far less, but it could be OS dependant

mvladic, 2016-12-08:
Yes, it takes more than 2 minutes. I tried again and upload blocked after exactly 2 minutes and 30 seconds.

facchinm, 2016-12-12:
Ok, so we reached a corner case which I never thought it was possible to encounter, my bad. 
@cmaglie @PaulStoffregen for your experience, what could be a safe value for MAXIMUM successful upload duration? 3 minutes? 5 minutes?

prasimix, 2016-12-12:
If 2 minutes is enough for 256KB, 4 minutes should be enough for 512KB :) 
Maybe you can link that limit to selected board.

commit 80981960 (2016-12-15): Kill programmer if upload takes more than 5 minutes

Fixes #5672
closed 2016-12-15 by facchinm