arduino/Arduino #5673: Simple Watchdog Support

mikaelpatel, 2016-12-03:
I would like to suggest an API to add simple watchdog support to the Arduino core. The default behavior could be to reset the board if locked up more than 5 seconds. The API could be implemented with the hardware watchdog or with a simple counter updated by the millis() ISR and timeout check in main(). Please see the below fork and latest commit for an implementation. 

https://github.com/mikaelpatel/Arduino/commit/80cad523962a53632c4c31cd6e7a8ec62a422bad

Cheers!

facchinm, 2016-12-05:
Hi @mikaelpatel ,
adding an official Watchdog API is a good idea, but your implementation (the linked one) is not a watchdog at all :slightly_smiling_face: If something goes really wrong, the "watchdog" will never be triggered, so it's useless. Also, since the behaviour of a watchdog is highly unexpected, activating it by default is a terrible idea for beginners (and also for experts who are trying to debug a particular piece of code). 
Instead, a good API similar to SAM one (https://github.com/arduino/Arduino/blob/master/hardware/arduino/sam/cores/arduino/watchdog.h) could be more than welcome

mikaelpatel, 2016-12-05:
@facchinm Thanks for the API link. I had missed that. It looks very much the same :). Is this also available for AVR based boards? Or should AVR wdt be used instead? 

I do not want to take your time further but I think you actually missed how the linked implementation works. To not work the millis() timer ISR has to break down. This covers a lot of the typical programming errors that new Arduino programmers do. Obviously it does not handle hardware failures, etc, which I think you indented with "really wrong". 

The idea behind this issue is that beginners need support from the boards on programming errors. A watchdog could help capture some of the error situations without requiring beginners to add any code. The default restart function could be a simple rapid blink function to indicate that the sketch locked up. 

Last but not least, this should capture when the heap is exhausted, i.e. malloc() returns NULL. This could reduce a lot of question on sketches that over-use the String class (which lacks a lot of error handling).

matthijskooijman, 2016-12-05:
I had a look at the code, it is indeed a watchdog implementation, but I see two problems with it:
 - It needs the timer ISR to work, so lockups caused by disabling interrupts will not be caught.
 - It resets by jumping to address 0, which does not reset hardware state and thus might cause all kinds of further unexpected behaviour.

Using the hardware watchdog (on AVR) makes more sense in this regard, since it solves both of these problems (but, due to the limited timeout values, results in a less portable API).

One advantage of this "software watchdog", is that there is more flexibility in providing feedback to the user that something went wrong, though the AVR watchdog interrupt can also provide the same.

Instead of adding this to the core API (with the portability issues and limitations resulting from that), I would rather encourage people to use a library that configures the real watchdog, which allows for more easily using the actual features of the platform.