arduino/Arduino #5674: IDE Timeouts while attempting to upload to Curie devices from different manufacturers

Bazmundi, 2016-12-04:
When trying to upload blink.ino the IDE reports:

```
Arduino: 1.6.12 (Windows 10), Board: "Arduino/Genuino 101"

Sketch uses 17,628 bytes (11%) of program storage space. Maximum is 155,648 bytes.
Starting download script...
Flashing is taking longer than expected
Try pressing MASTER_RESET button
ERROR: Timed out waiting for Arduino 101 on COM7
ERROR: Timed out waiting for Arduino 101 on COM7
```

1. Against both DFRobot CurieNano and SparkFun Genuino 101 (so two different device manufacturers)
2. On two windows 10 machines (Desktop PC and on a Laptop, on all USB ports on both machines, using various cables including new data cables purchased to ensure correct cable type)
3. With fresh install of 1.6.12 IDE (and 1.6.9 as part of trouble shooting)
4. With Intel Curie 1.7 core installed in IDE (other cores also fail but with different error messages)
5. Drivers installed and reported in Device Manager as working (various disabling and manual installing of drivers subsequently applied during trouble shooting)
6. Board selected "Arduino/Genuino 101"
7. Genuino 101 shown on Port selection
8. Various programmers selected during trouble shooting
9. When requested IDE can provide board info for NanoCurie and for Genuino 101

Same problem has been reported by others on Linux.  It appears to be IDE problem.

Response from Intel was IDE the likely source of problem.

facchinm, 2016-12-05:
Keeping track of this issue on the forum, so closing this one, thank you

closed 2016-12-05 by facchinm

Bazmundi, 2016-12-05:
No joy on forums though as they are tailored for generic conversation and not issue tracking.  

A fallacy then as the issue is not closed as solution has not been wrought.

We agree it's an issue I assume.

Looks more an more like a problem with devices in any event as they can't be seen with df-util.

Neither Intel or DFRobot are admitting to duds in chips or boards respectively.  I haven't approached SparkFun as I didn't buy direct and so "Gray" rules apply.

Bazmundi, 2016-12-05:
Definitely a dud device.  I just tried to upload firmware using:

```
Select Tools >> Programmer >> Arduino 101 Firmware Updater
Select Tools >> Burn Bootloader
```

Got the following as a response:

```
Arduino: 1.6.13 (Windows 10), Board: "Arduino/Genuino 101"

Reset the board before proceeding...
Error while burning bootloader.
```

So as I have seen this on two different boards from two different manufacturers it is pointing at Intel and the Curie chip, unless there is a common mode failure in the reference design that both DFRobotics and SparkFun have adopted in the board designs.

facchinm, 2016-12-06:
@Bazmundi ,
check your PM messagebox on the forum :wink:

sjlawson, 2017-04-30:
I had the same issue with a 101 I brought of the shelf from Fry's in April 2017. On a whim, I downgraded the board version from 2.02 to 1.07 and it worked.

rlmccormick, 2017-05-09:
Having the same issue with a board I just bought today at Fry's.  What do you mean by: "I downgraded the board version from 2.02 to 1.07 and it worked." ?

sjlawsonAL, 2017-05-10:
I mean I had to downgrade the driver version for the board from 2.02 to 1.07 in the Board Manager.
I'll have to go back and check and make sure that was actually necessary, because it was a couple of things. One definite is that I have to press the Master Reset button before uploading new code, otherwise it never works.  I will verify the version when I get home from work. A colleague of mine pointed out that the 'programmer' setting in the IDE might also not be 100% compatible, since the 101 itself is not %100 compatible with Arduino UNO, which is probably the cause for most of the flakiness.

Bazmundi, 2017-05-11:
You people either have more time on you hands than me OR you think its okay
for people to sell you flaky widgets.  It isn't okay.

vtsselvan1995, 2017-06-24:
BAZMUNDI... hope this reply is too late, but still i installed another version of 101 in the boards. Try that option

Bazmundi, 2017-06-25:
I am surprised people don't understand that my curie boards where not seen
by 3 different computers. They weren't seen by the arduino ide using
various core versions. They were not seen by the intel curie factory flash
tool. They could NOT be re-flashed. The software has to detect the boards
to flash them.

They are long in the garbage.

Rafmigz, 2020-11-27:
Hi! I was really persistent to look for ways on how to fix the Arduino/Genuino 101 Timeout Error, I searched different forums and read a lot of suggestions on how to fix the error, but still all failed, I know some of the 101 users already bricked their 101 or other just left it behind.
We are all aware that the Arduino 101 is already discontinued and even support is already stopped, Hence this little but powerful microcontroller is still out there and probably used by the some. Anyhow, to be straight to the point I finally I discovered the fix for the error.
Notice that when you insert your 101 to the USB plug it will read and install the device, checking it in the device manager will show the Arduino/Genuino 101 in "Other Devices" and a new COMPort (COMX) is added, then you install the Arduino IDE (regardless of the version), then install the Arduino 101 Board thru "Board Manager", the manager will install the Arduino 101 but it will not notify you that the driver for Arduino 101 is not installed due to error. So in short we are assuming that the driver for Arduino 101 is successfully installed. This is where the error lies. The software driver "dpinst-amd64" is not digitally signed and Windows is blocking to install unsigned drivers.
So what's the fix?
The fix here is still to install the "dpinst-amd64" since it is the driver needed to recognize your 101, the trick here is you need to install it in other way, not in your regular PC operation. You need to access the "disable signature enforcement" under the startup options (not safe mode). This startup mode will enable you to install unsigned drivers (I trust this Intel Curie driver since it was downloaded with the board). After installation restart your PC to its regular startup and plug-in again your Arduino/Genuino 101, this time the PC will now recognize the microcontroller and you can now upload sketches without the time out error.

For a step by step procedure:
1. Access Update and Security>Recovery>Advanced Startup
2. Your computer will eventually restart and will bring you to the Advanced Startup, access Troubleshoot>Startup Settings. Your PC will restart again and options for the startup will display, choose number 7 "disable signature enforcement"
3. Your PC will start. Locate "dpinst-amd64.exe" in C:>Users>YOUR USERNAME>AppData>Local>Arduino15>packages>Intel>hardware>arc32>2.0.4(latest version)>drivers>dpinst-amd64
4. During installation a message will prompt if you want to install the unsigned driver, just click install.
5. After installation restart your PC and your Arduino/Genuino 101 is good to go.
6. To confirm that your Arduino/Genuino 101 is installed go to Device Manager and you will see that the Arduino/Genuino 101 is not anymore in the Other Devices.

I hope this will help. :D I will try to make a video about this fix. :D

Thanks!

Regards, 

Raf

Bazmundi, 2020-12-01:
I got around the problem by throwing the devices in the bin.

jha-prateek, 2020-12-08:
> I mean I had to downgrade the driver version for the board from 2.02 to 1.07 in the Board Manager.
> I'll have to go back and check and make sure that was actually necessary, because it was a couple [...]

Did not work for me.