arduino/Arduino #5675: IDE ignores LOCALAPPDATA environment variable

iddq, 2016-12-04:
Sorry if I'm wrong but I think IDE ignores LOCALAPPDATA environment variable

arduino.bat:
```
set LOCALAPPDATA="C:\arduino-1.6.13\AppData\Local"
arduino.exe
```

matthijskooijman, 2016-12-04:
Do you have any (official) references on how this environment variable is intended to work? Should applications read it directly, or should they use some Windows API to find the appdata folder?

iddq, 2016-12-04:
from revisions.txt:
* Windows: IDE stores its stuff under %LOCALAPPDATA% instead of %APPDATA% (which was a roaming folder). %APPDATA% content is automatically migrated to new location

iddq, 2016-12-04:
can you please support the configurable settingsFolder?

facchinm, 2016-12-05:
Hi @iddq ,
`%LOCALAPPDATA%` is a special Windows folder, not a user environment variable. 

Making the "configuration" folder user-configurable could bring little to no improvement (cores are invisible unless you want to modify them, and in that case you should use a version-controlled folder before modifying) at the cost of adding a lot of complexity; for example, that folder contains the `preference` file which specifies everything about the IDE behaviour, it must be there, so the folder will be created and used anyway :smile: 

Making the IDE "[portable](https://www.arduino.cc/en/Guide/PortableIDE)" and replacing the portable folder with a symlink to a user defined folder is probably what you are looking for.
Feel free to reopen the issue if you feel that I missed something, thanks!

closed 2016-12-05 by facchinm

iddq, 2016-12-05:
yes you are right, I've made some modification in the core lib and if I upgrade the IDE and the core libs I don't want that to be overwritten. Anyways it would be a nice feature in case of portable IDE and if someone want to have multiple versions on his pc.