arduino/Arduino #5679: Upload Error on MKR1000

HKOLWD, 2016-12-04:
The IDE fails to upload saying it cannot find anything on the COM port on Windows. Similar error on Mac.

It seems this error sometimes appears:

processing.app.debug.RunnerException
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:160)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:78)
	at processing.app.SketchController.upload(SketchController.java:709)
	at processing.app.SketchController.exportApplet(SketchController.java:682)
	at processing.app.Editor$DefaultExportHandler.run(Editor.java:2189)
	at java.lang.Thread.run(Thread.java:745)
Caused by: processing.app.SerialException: Error touching serial port 'COM5'.
	at processing.app.Serial.touchForCDCReset(Serial.java:87)
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:144)
	... 5 more
Caused by: jssc.SerialPortException: Port name - COM5; Method name - openPort(); Exception type - Port busy.
	at jssc.SerialPort.openPort(SerialPort.java:164)
	at processing.app.Serial.touchForCDCReset(Serial.java:81)
	... 6 more

facchinm, 2016-12-05:
Hi @HKOLWD, could you add to the report the IDE version and the SAMD core version? This would greatly help our debugging. 
Anyway, the problem is that your PC is unable to "touch" the serial port to restart the board in bootloader mode because the port is locked by another process. Do you have any serial terminal running (like putty)? If you manually enter bootloader mode (by double clicking the RESET button) can you upload correctly?

HKOLWD, 2016-12-05:
Hi @facchinm. Thanks for the fast response. I have to admit I am a bit of a beginner in all of this, so I hope you have some patience with me :)

The IDE version is 1.6.13. I am not sure what SAMD core it is. Where can I find this? 

I have the feeling the Genuino itself is keeping the COM port locked. 
1) The uploads were OK on both Mac and Windows all the time. Suddenly, on the Mac, is stopped uploading (it will simply hang ending in the message that it cannot find a device)
2) After switching to windows, the problem persisted (while uploads from windows had been going fine previously as well), with the error above as a result.

I have tried double tapping the reset button, but it had no effect. Not when the genuino was running, nor when I took power supply off. The upload problem kept persisting.

I tried swapping from COM5 to COM100 on Windows, but then the same error will simply occur for the COM100 port.

Hope this helps...

facchinm, 2016-12-05:
From your report it appears that you loaded a sketch which hangs the USB connection.
Make sure you are entering bootloader mode by double clicking (quickly) the reset button. Select the correct serial port and click "Get board Info" under "tools" menu, it should report ```
BN: Arduino/Genuino MKR1000
VID: 2341
PID: 004E
SN....
```

If you see PID: 804E you are not in bootloader mode and you must repeat the double click. Once this is ok, try uploading a Blink without disconnecting the board. Let me know if it works!

HKOLWD, 2016-12-05:
After the double tap, and following the instructions on the Mac, the response was this:
BN: Arduino/Genuino MKR1000
VID: 2341
PID: 024E
SN: Upload any sketch to obtain it

I tried uploading the blink example (which failed yesterday) and it worked! 

I suppose I will have to take a new punch at my attempt to make a sketch for the LED display :)

HKOLWD, 2016-12-05:
Just out of interest, if I can ask here. What can be the cause of hanging the USB connection (in a sketch)?

facchinm, 2016-12-06:
Since the USB is IRQ based, completely disabling the interrupts with `noInterrupts()` can indeed make the USB nonfunctional. A memory overflow can trigger strange behaviours too. BTW, glad you got it working again :smile: !

closed 2016-12-06 by facchinm

Kevinjconnorsjr, 2019-08-13:
> From your report it appears that you loaded a sketch which hangs the USB connection.
> Make sure you are entering bootloader mode by double clicking (quickly) the reset button. Select the correct [...]

What would one do if they ran blink after and it built the same error message that it was receiving on the other sketches?

Kevinjconnorsjr, 2019-08-13:
So one method that I tried that worked was switching to a different USB port which got it to work right away, didn't need to reset it. I tried all the other previous steps without success, but using a completely different USB and COM port seemed to do the trick for me, not sure how repeatable that will be for others though.

Good Luck!

SimonMerrett, 2020-03-23:
I'm not sure what worked when I had this issue. One thing I think could have been a problem was that there was a bluetooth com port taking the same number as the board was being allocated in the IDE. I used com_name_arbiter_setter to sort but no joy, so I disabled the bluetooth in Device Manager on Win10. I also threw in some re-burning of the bootloader and fuse setting for good measure, hence not sure quite what combination fixed things! Maybe some tips there for others though.