arduino/Arduino #5691: avr-g++ file

arun1944, 2016-12-09:
I am trying to compile a simple "BLINK" sketch for Arduino UNO board, using Arduino IDE v1.6.13 on my Win 8.1 OS.
But while compiling, the output window shows an error message 
C:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\4.9.2-atmel3.5.3-arduino2/bin/avr-g++": file does not exist
Error compiling for board Arduino/Genuino Uno.

How do I resolve this issue, any suggestions please ?
-arun karkare

dlabun, 2016-12-11:
Sounds like your installation is corrupt... You may want to try removing and reinstalling Arduino

facchinm, 2016-12-12:
Before starting the uninstall/reinstall dance, try opening the Board Manager and uninstall the AVR core. If everything goes well the label should become "BUILTIN" and the IDE should start using the builtin core again (with its related tools)

facchinm, 2016-12-13:
@arun1944 did it solve your issue? :smile:

arun1944, 2016-12-13:
I had to delete the "C\Users\...Arduino15" file, uninstall the Arduino IDE
program, and then re-install 1.6.13 version.
Now it is working fine.
Anyway Thanks for ur concern
-Arun

closed 2016-12-13 by facchinm

Nibbels, 2017-08-22:
xec: "C:\\Users\\USERNAME\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\4.9.2-atmel3.5.4-arduino2/bin/avr-g++": file does not exist
Fehler beim Kompilieren für das Board Arduino/Genuino Mega or Mega 2560.

This happens when running a suggested "Board update" while compiling something. Then the update doesnt work and the board is unusable.

facchinm, 2017-09-04:
@Nibbels probably on the first round the extraction failed but the folder structure (and relevant files) where in the correct place, so every other installation would pick theme as the correct core. Uninstalling und reinstalling the core from board manager does the trick as you discovered :wink:

binmulyadin, 2017-09-15:
My problem solved as suggested by Nibbels, thank you

Steve92169, 2018-01-20:
I had the same error message on a Mega 2560 R3 using IDE 1.8.5 and AVR Boards 1.6.20. I installed AVR Boards 1.6.19, and that solved the problem.  Re-installing 1.6.20 failed, so I went back to 1.6.19. It seems to work fine.

davidm-public, 2018-02-23:
@facchinm  worked for me!
thanks

adumont, 2018-07-23:
@Steve92169 solution above worked for me.

Gurubasavaraj003, 2018-09-02:
same problem but this time also facing the same issue any solution plz

irfanctn, 2018-11-11:
you change arduino's home directory to the root directory, long folder names causes this error

Aaron-Roar, 2021-01-09:
I am using arduino-cli on Alpine Linux and I received this error. After some investigation I noticed it was dynamically linked when running
`Sat Jan  9 06:03:49 UTC 2021> file ~/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++`
with output
`/root/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++: ELF 64-bit LSB executable, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, for GNU/Linux 2.6.26, BuildID[sha1]=a96cf70b452ab1acb76a923f4aba04eeade2a926, stripped`

I was missing dependencies to handle this, specifically /lib64/ld-linux-x86-64.so.2
This dependency can be acquired from the alpine repository https://pkgs.alpinelinux.org/contents?file=ld-linux-x86-64.so.2
`apk add gcompat`

after acquiring such package I am able to compile

`Sat Jan  9 06:09:24 UTC 2021> ~/arduino-cli compile -b arduino:avr:uno`
with output
`Sketch uses 924 bytes (2%) of program storage space. Maximum is 32256 bytes.
Global variables use 9 bytes (0%) of dynamic memory, leaving 2039 bytes for local variables. Maximum is 2048 bytes.
`

Mayoogh, 2021-06-04:
Downgrading Arduino AVR Boards from `1.8.3` to `1.8.1` from the board manager fixed the issue for me

giulionyakunga, 2021-12-03:
This issue occurred to after updating Arduino AVR Boards from older version to version 1.8.4, to fix this I had to downgrade  Arduino AVR Boards to version 1.8.3, so I recommend you to install Arduino AVR Boards version 1.8.3 through this simple steps, go to Tools -> Boards -> Board Manager -> Install "Arduino AVR Boards (1.8.3)"
Good luck...

rovesoul, 2022-08-05:
> This issue occurred to after updating Arduino AVR Boards from older version to version 1.8.4, to fix this I had to downgrade Arduino AVR Boards to version 1.8.3, so I recommend you to install [...]

thanks,your answer solve my problem.