arduino/Arduino #5692: www.arduino.cc/en/Reference/FunctionDeclaration.html

AndrewMac-CT, 2016-12-09:
In "Anatomy of a C Function". As a newbie programmer (68years old) I struggled to get Functions to work for over 4 weeks. Secret was the "curved braces required"  ( ) Even if no parameters are passed. just like "curly braces required" ....Aaaagh.

akash73, 2017-01-16:
@AndrewMac-CT thanks for notice.
can you share the link of the page?
thanks

AndrewMac-CT, 2017-01-16:
Hi Andrea,

the page is:

https://www.arduino.cc/en/Reference/FunctionDeclaration

If a Void function ie nothing is returned and nothing is input then the 
the function is a void function say

x=myFunctionName();       // "(" and ")" brackets plus ";" must be given 
even if no parameters are passed in or out

many Thanks and kind regards.

Andrew Macgregor

Cape Town

+27716885039

akash73, 2017-02-10:
Hi @AndrewMac-CT ,
we added a paragraph at the end.
thanks.

closed 2017-02-10 by akash73