arduino/Arduino #5694: Boards Manager doesn't process additional URLs when main package index is not accessible

shadson, 2016-12-09:
Hello,

Problem:
if Boards Manager fails to download main package index - it doesn't try to access additional boards URLs , even if they are located not on 'downloads.arduino.cc'.

I try to add ESP8266 support to Arduino IDE, so I put URL of additional board index into 'Additional Boards Manager URLs' (IDE->File->Settings).
Then I start Boards Manager, but 'downloads.arduino.cc' is not accessible now, so IDE can't download main package index. The data from additional index file are not showed in the list also.

Tested on 1.6.8 and 1.6.13.

Workaround:
- download and start any simple http web server on port 80
- add line '127.0.0.1 downloads.arduino.cc' into 'hosts' file
- download 'package_index.json' and 'package_index.json.sig' from archive.org
- put them into 'packages' folder of web server files folder
- start Arduino IDE, start Boards Manager
- wait until main packages index will be processed
- install additional board support from the last line in Boards Manager

facchinm, 2016-12-12:
Should be solved right now, downloads.arduino.cc is up, however thanks a lot for providing the workaround!

closed 2016-12-12 by facchinm

alshorin, 2017-01-21:
or easier:
Workaround:

    <<download and start any simple http web server on port 80>>  unnecessary!
    add line 'file:////home/<yourName>/tmp/package_index.json' into 'hosts' file
    download 'package_index.json'  from archive.org
    put them into 'tmp' folder 
    start Arduino IDE, start Boards Manager
    wait until main packages index will be processed
    install additional board support from the last line in Boards Manager