arduino/Arduino #5696: TFT Library Compile Error on Zero and MKR1000

tonywhitfort, 2016-12-09:
I get the following error when compiling any of the TFT/Arduino examples for either Zero or MKR1000
I do not get this error when compiling for UNO and similar
There are also some warnings that can probably be ignored.

Arduino IDE version 1.6.13 running on mac 

/Applications/Arduino.app/Contents/Java/libraries/TFT/src/utility/Adafruit_ST7735.cpp: In member function 'void Adafruit_ST7735::commonInit(const uint8_t*)':
/Applications/Arduino.app/Contents/Java/libraries/TFT/src/utility/Adafruit_ST7735.cpp:349:13: error: cannot convert 'volatile uint32_t* {aka volatile long unsigned int*}' to 'volatile uint8_t* {aka volatile unsigned char*}' in assignment
   csport    = portOutputRegister(digitalPinToPort(_cs));

AND THE SAME FOR THREE SIMILAR LINES 
rsport      = portOutputRegister(digitalPinToPort(_rs));
clkport    = portOutputRegister(digitalPinToPort(_sclk));
dataport  = portOutputRegister(digitalPinToPort(_sid));

dlabun, 2016-12-11:
I quickly looked at the Adafruit library and it does not appear it is written to work with the Zero or MKR1000. This is not the proper place to ask for support, this is really an issue with the Adafruit library.

q2dg, 2016-12-11:
Well...the fact is TFT library was retired due to Arduino war and Zero/MKR1000 boards came later. So now war has ended TFT library has returned. The question, however is: how much effort maintaining an improving this library will assume Arduino team?

tonywhitfort, 2016-12-11:
Thanks for the comments.
I have downloaded the latest version of Adafruit's ST7735 library and that compiles and runs OK, so I will use that.
I guess the question is if the the Arduino IDE should include TFT examples that work with Zero and MKR1000

per1234, 2017-07-06:
>The question, however is: how much effort maintaining and improving this library will assume Arduino team?

At the very least, if the problem with SAMD compatibility is not going to be fixed then the `architectures` value in [library.properties](https://github.com/arduino-libraries/TFT/blob/master/library.properties#L9) needs to be changed.

Meins321, 2020-06-26:
Just using the arduino zero with
#include <TFT.h>  // include TFT
will give this:
In file included from K:\arduino-1.8.12\portable\packages\arduino\hardware\samd\1.8.6\cores\arduino/delay.h:23:0,

                 from K:\arduino-1.8.12\portable\packages\arduino\hardware\samd\1.8.6\cores\arduino/Arduino.h:81,

                 from K:\arduino-1.8.12\libraries\TFT\src\utility\Adafruit_ST7735.h:23,

                 from K:\arduino-1.8.12\libraries\TFT\src\utility\Adafruit_ST7735.cpp:19:

K:\arduino-1.8.12\libraries\TFT\src\utility\Adafruit_ST7735.cpp: In member function 'void Adafruit_ST7735::commonInit(const uint8_t*)':

K:\arduino-1.8.12\portable\packages\arduino\hardware\samd\1.8.6\variants\arduino_zero/variant.h:68:55: error: cannot convert 'volatile uint32_t* {aka volatile long unsigned int*}' to 'volatile uint8_t* {aka volatile unsigned char*}' in assignment

 #define portOutputRegister(port)   ( &(port->OUT.reg) )

                                                       ^

K:\arduino-1.8.12\libraries\TFT\src\utility\Adafruit_ST7735.cpp:349:15: note: in expansion of macro 'portOutputRegister'

   csport    = portOutputRegister(digitalPinToPort(_cs));

               ^~~~~~~~~~~~~~~~~~

K:\arduino-1.8.12\portable\packages\arduino\hardware\samd\1.8.6\variants\arduino_zero/variant.h:68:55: error: cannot convert 'volatile uint32_t* {aka volatile long unsigned int*}' to 'volatile uint8_t* {aka volatile unsigned char*}' in assignment

 #define portOutputRegister(port)   ( &(port->OUT.reg) )

                                                       ^

K:\arduino-1.8.12\libraries\TFT\src\utility\Adafruit_ST7735.cpp:351:15: note: in expansion of macro 'portOutputRegister'

   rsport    = portOutputRegister(digitalPinToPort(_rs));

               ^~~~~~~~~~~~~~~~~~

K:\arduino-1.8.12\portable\packages\arduino\hardware\samd\1.8.6\variants\arduino_zero/variant.h:68:55: error: cannot convert 'volatile uint32_t* {aka volatile long unsigned int*}' to 'volatile uint8_t* {aka volatile unsigned char*}' in assignment

 #define portOutputRegister(port)   ( &(port->OUT.reg) )

                                                       ^

K:\arduino-1.8.12\libraries\TFT\src\utility\Adafruit_ST7735.cpp:370:19: note: in expansion of macro 'portOutputRegister'

     clkport     = portOutputRegister(digitalPinToPort(_sclk));

                   ^~~~~~~~~~~~~~~~~~

K:\arduino-1.8.12\portable\packages\arduino\hardware\samd\1.8.6\variants\arduino_zero/variant.h:68:55: error: cannot convert 'volatile uint32_t* {aka volatile long unsigned int*}' to 'volatile uint8_t* {aka volatile unsigned char*}' in assignment

 #define portOutputRegister(port)   ( &(port->OUT.reg) )

                                                       ^

K:\arduino-1.8.12\libraries\TFT\src\utility\Adafruit_ST7735.cpp:372:19: note: in expansion of macro 'portOutputRegister'

     dataport    = portOutputRegister(digitalPinToPort(_sid));

                   ^~~~~~~~~~~~~~~~~~

exit status 1
Fehler beim Kompilieren für das Board Arduino Zero (Native USB Port).

Still not resolved in 2020 with original board+ide and latest adafruit gfx library.. what a shame

ATmel91, 2020-10-12:
I am having the exact same issue. Its annoying as hell.

ubidefeo, 2020-10-12:
@ATmel91 
try another library for that display
I compile successfully with Adafruit-ST7735 for any SAMD21 board including the Zero

per1234, 2021-03-28:
Transferred to https://github.com/arduino-libraries/TFT/issues/24

closed 2021-03-28 by per1234