arduino/Arduino #5703: MKR1000 wifi hangs with Wifi101 0.11.1

bcraigie, 2016-12-11:
I've spent all day trying to figure out why my MKR1000 has been hanging a few seconds after a Wifi.begin instruction.  Sometimes straight away after, and sometimes when I try to send data to a Thingspeak server a few seconds later.  It seems to be random whether it hangs immediately after the connection succeeds or when the first data is transmitted.  Data is set to upload every 30 seconds on a timer, so it's not that it is going too fast.

I was using Wifi 101 library version 0.11.1.  When it hung the Arduino, it also became un-programmable and I had to double-press the reset button to make it programmable again.  This also caused the com port to change its number.

Eventually in despair, I downgraded the WiFi 101 library to version 0.10.0 and all is well again and the com port number is no longer getting changed.  This is all without changing any of my code.

It would _appear_ that somewhere in the 0.11.1 library the code is going into an infinite loop and stopping all external responses to the MKR1000.

If it is relevant, I also have an HC-05 bluetooth module, a DHT11 and an F-RAM on board.  I was using the Bluetooth to see my debugging statements because nothing was emerging on the serial monitor.

In researching this, I have seen many reports of the WiFi101 library hanging - it seems to be an ongoing problem.  Anyway, 0.10.0 seems to be reliable on the MKR1000.

q2dg, 2016-12-11:
Maybe reading this (https://github.com/arduino-libraries/WiFi101/issues/117) is worthy for you

bcraigie, 2016-12-11:
Ooh.  Thank you. :-)  I have now subscribed to that thread.

sandeepmistry, 2016-12-12:
Hi @bcraigie,

Could you please share a minimal sketch that we can use to reproduce the issue?

sandeepmistry, 2016-12-12:
Closing this, discussion has been moved to https://github.com/arduino-libraries/WiFi101/issues/117.

closed 2016-12-12 by sandeepmistry