arduino/Arduino #5709: Hourly build has not been updated since 2016-11-25

per1234, 2016-12-12:
Latest hourly build available from https://www.arduino.cc/en/Main/Software is Arduino IDE 1.6.14 2016/11/25 05:33 but there have been several commits since that time that don't appear to be included in the hourly build(https://github.com/arduino/Arduino/issues/5550 still occurs).

oqibidipo, 2016-12-12:
Only the Windows version is that old, Mac and Linux versions are dated 2016/12/07.

facchinm, 2016-12-12:
Thanks thanks thanks a lot for reporting this. Jenkins was failing the build but no mail was sent so we didn't catch it immediately. Should be solved by https://github.com/arduino/Arduino/commit/83f9ed7e0acae26e4c26fcf286fa0258f2fd1594 . Thanks again!

closed 2016-12-12 by facchinm

oqibidipo, 2016-12-12:
Looks good:
```
arduino-nightly-linux32.tar.xz  2016/12/12 10:31
arduino-nightly-linux64.tar.xz  2016/12/12 10:26
arduino-nightly-linuxarm.tar.xz 2016/12/12 10:19
arduino-nightly-macosx.zip      2016/12/12 11:12
arduino-nightly-windows.zip     2016/12/12 10:16
```