arduino/Arduino #5710: Programmer Gemma v2

mwhuss, 2016-12-12:


mwhuss, 2016-12-13:
If it makes a difference this was for the "Adafruit GEMMA v2" board.

mwhuss, 2016-12-14:
Doing more research, it does look like the Adafruit Gemma and the Arduino Gemma are different. I'm currently using `v1.6.13` which is supposed to have the Adafruit boards available in the board manager according to [this article](https://learn.adafruit.com/adding-custom-boards-to-the-arduino-v1-6-board-manager).

Alternatively I was able to get them to show up by adding this as an "Additional Boards Manager URL".

https://adafruit.github.io/arduino-board-index/package_adafruit_index.json

sandeepmistry, 2017-02-16:
The article you linked to: https://learn.adafruit.com/adding-custom-boards-to-the-arduino-v1-6-board-manager has a warning:
> This guide is no longer accurate with the release of version 1.6.4 of the Arduino IDE. Custom third party board support has been officially added in v1.6.4+.

So adding the board manager URL is the way to go. I'm closing this as resolved.

closed 2017-02-16 by sandeepmistry