arduino/Arduino #5712: Listening for compiler updates from tool

tomneutens, 2016-12-12:
I would like to show the compile/upload progress in my tool. As far as i can see in the source code the editor.status::progressUpdate method is passed to the compiler build() method in the SketchController class. Consequently, there is no way of adding my own listener to the update process. 

I can see two options to make the behaviour i want to implement possible:
1. quick and dirty: provide a getter for the "JProgressBar progressBar" in the EditorStatus class so i can listen for updates of the JProgressBar.

2. make it possible to add listerners to a listenerlist in the SketchController class so mulitple objects can listen for the build updates.

Is this something which could bed added?

facchinm, 2016-12-12:
Hi @tomneutens , such an API could be truly useful for 3rd party tools developers (I believe). Can you propose a PR so we can discuss on that? Thanks a lot!

per1234, 2017-07-06:
Closing as solved by https://github.com/arduino/Arduino/pull/5716.

If this is not the case please let me know and I'll reopen.

closed 2017-07-06 by per1234