arduino/Arduino #5718: NullPointerException using CLI --port parameter

SukkoPera, 2016-12-14:
I'm trying to upload a sketch via the CLI interface:

`arduino --pref build.path=/tmp/arduproj/MegaDrivePlusPlus --preserve-temp-files --upload ~/Development/arduino/MegaDrivePlusPlus/MegaDrivePlusPlus.ino --port /dev/ttyACM0 --board arduino:avr:uno --verbose-upload`

But uploading fails with the following:
```
Picked up JAVA_TOOL_OPTIONS: 
Sto caricando la configurazione...
java.lang.NullPointerException
	at processing.app.BaseNoGui.selectSerialPort(BaseNoGui.java:1126)
	at processing.app.helpers.CommandlineParser.parseArgumentsPhase1(CommandlineParser.java:143)
	at processing.app.Base.<init>(Base.java:273)
	at processing.app.Base.guardedMain(Base.java:219)
	at processing.app.Base.main(Base.java:132)
```

OS is Linux and Arduino version is 1.6.12.

per1234, 2016-12-15:
Duplicate of https://github.com/arduino/Arduino/issues/5413

You're reporting an issue that was fixed 2.5 months ago. Please always download the [hourly build](http://www.arduino.cc/en/Main/Software#hourly) to check if the issue has already been fixed and do a search of the issues(including closed issues) to be sure it hasn't already been reported before submitting an issue.

SukkoPera, 2016-12-15:
Sorry for that. I did a search, but I guess I didn't use good keywords. Thanks.

closed 2016-12-15 by SukkoPera