arduino/Arduino #5719: https://github.com/CsCrazy85/Arduino/releases/tag/1.0.0

CsCrazy85, 2016-12-15:
DatavisionLCD library 1.0.0

per1234, 2016-12-15:
Are you requesting your library to be added to Library Manager. If so you should review https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ:
>Your repo is ok if its root folder contains file library.properties and the rest of the library source code. If your library is stored into a subfolder, your repo it's not ok and we can't include your library in the list.

facchinm, 2016-12-15:
Hi @CsCrazy85 ,
as @per1234 pointed out, your repo is not in the right form to be included in library manager. You need to move all the files from DatavisionLCD folder to the root of the repo.
No need to open another issue if you fix these things, so I closed #5720

facchinm, 2016-12-16:
Library added, you need to modify https://github.com/CsCrazy85/DatavisionLCD/blob/master/library.properties#L2 to match the tag to make it appear in library manager (modify that line to 1.0.2 and tag 1.0.2 to do so).
Thanks!

closed 2016-12-16 by facchinm

CsCrazy85, 2016-12-16:
Okay! I will do so! Thank you!