arduino/Arduino #5722: AJAX/HAR Error:0 during Arduino Configuration Wizard OS ver 2.0.1

ronpod, 2016-12-15:
An error (AJAX/HAR Error:0) occurs during the configuration of the Yun for WiFi. The Arduino Configuration Wizard inputs all data from user then during Save and Reconfiguration the error occurs between announcements of "Warning: Off-line" and "Warning:On-line". Every once in awhile 100% upload occurs but the error does not allow entry to 192.168.240.1. The Yun does not join the network.

facchinm, 2016-12-15:
Hi @ronpod ,
could you specify the OS version of your Yun? And what do you mean with "Arduino Configuration Wizard" ? Thanks

ronpod, 2016-12-15:
Hi @facchinm ,
I currently write my sketches on 1.6.14 Hourly Build 2016/11/25 05:33. To the point of your question; the Yun is Arduino OS Version 2.0.1-arduino-os. The OS.js project is by Anders Evenrud. The steps of configuring the Yun begin with a factory reset of 30 seconds of wlan reset button (ensuring that the blue led is flashing the whole time). Then connect my laptop wifi to the Yun and open 192.168.240.1 with psswd "arduino". The Arduino Configuration Wizard opens with a dialog box that looks different from all the multiple demos online. But the same user data is required. Here Luci is available with a few clicks. Once user data is input then the Save and Restart button is clicked. Little warning announcements are displayed as written above except that AJAX/XHR Error:0 is displayed. Sometimes the Wizard self connects the laptop back on to the network, but sometimes the Wizard asks for the user to reconnect to the network (I have reset this many times). Then when trying to sign back on to 192.168.240.1 with any psswd an AJAX/XHR Error:0 is produced. The Yun does not join the network but  the laptop does wifi connect through the sketch. I picked up the Yun from Sparkfun and couple of days ago and was told that it was loaded with the latest OS. All wifi devices are within 18 inches of each other and the signal strength is 100%. 
Thanks for your help!

ronpod, 2016-12-15:
Hi @facchinm ,
I currently write my sketches on 1.6.14 Hourly Build 2016/11/25 05:33. To the point of your question; the Yun is Arduino OS Version 2.0.1-arduino-os. The OS.js project is by Anders Evenrud. The steps of configuring the Yun begin with a factory reset of 30 seconds of wlan reset button (ensuring that the blue led is flashing the whole time). Then connect my laptop wifi to the Yun and open 192.168.240.1 with psswd "arduino". The Arduino Configuration Wizard opens with a dialog box that looks different from all the multiple demos online. But the same user data is required. Here Luci is available with a few clicks. Once user data is input then the Save and Restart button is clicked. Little warning announcements are displayed as written above except that AJAX/XHR Error:0 is displayed. Sometimes the Wizard self connects the laptop back on to the network, but sometimes the Wizard asks for the user to reconnect to the network (I have reset this many times). Then when trying to sign back on to 192.168.240.1 with any psswd an AJAX/XHR Error:0 is produced. The Yun does not join the network but the laptop does wifi connect through the sketch. I picked up the Yun from Sparkfun and couple of days ago and was told that it was loaded with the latest OS. All wifi devices are within 18 inches of each other and the signal strength is 100%.
Thanks for your help!

Sorry for the double post - this is my first time...

facchinm, 2016-12-15:
No problem for the double post :smile: We are not the maintainers of ArduinoOS, so you'll find better answers on the official repo https://github.com/arduino-org/Arduino-OS . I'm closing this one, please reopen the bug there :wink: Happy hacking!

closed 2016-12-15 by facchinm

Liam1947, 2017-04-10:
Hi@ronpod,

Did you resolve this Yun problem?  If so, how?

Many thanks