arduino/Arduino #5724: Simple Nokia5110 LCD library to print strings on screen

nitins11, 2016-12-15:
Hi,

Submitting a simple Nokia5110 LCD library to print strings easily on the screen. None of the current LCD libraries or example codes on internet provide an easy solution to do so for Nokia5110 LCD. Below is the repository link:

https://github.com/nitins11/Arduino/tree/master/libraries/Nokia5110Lib

Happy to answer if any questions.

Thanks
Nitin

cmaglie, 2016-12-21:
I've added the library to the indexer, BTW the tag 0.1 has a top library folder:

https://github.com/nitins11/Nokia5110LCD/tree/v0.1

instead your `master` branch has the correct layout (all files on the root of the git repo).
To fix this I suggest you to delete the tag `0.1` and redo it on master branch. Once done the library will be automatically published within an hour.

closed 2016-12-21 by cmaglie

nitins11, 2016-12-21:
Thanks Christian! I have re-created the tag v0.1 on the master now.

Let me know if there is any pending action on my side.

Thanks
Nitin