arduino/Arduino #5732: Revert Subtract one from USB_EP_SIZE in USB_SendSpace

NicoHood, 2016-12-16:
https://github.com/arduino/Arduino/pull/4864 introduced a bug that rawhid and the gamepad is not working anymore. This was reported by a lot of users now and should be reverted. There are lots of others suggestions to fix this bug, please go another way. This is definitely not the correct way and breaks multiple USB modules. The core should not try to fix CDC module bugs.

More details in the pr above and here: https://github.com/NicoHood/HID/issues/91

NicoHood, 2017-01-03:
@facchinm @cmaglie People constantly contact me about this issue and I need to tell them to download the very old IDE 1.6.8 or 1.6.9 to fix this issue. Please revert the commit and then think of another fix.

mnhpias, 2017-03-10:
I'm unable to work RawHID.write ,
I changed \libraries\HID\src\SingleReport\RawHID.h in line 50 from #define RAWHID_TX_SIZE RAWHID_SIZE to #define RAWHID_TX_SIZE 63 , then I sent 50 bytes long uint8_t throgh RawHID.write , but pc didn't receive anything . pls help me to make RawHID.write working .

PR #5802 (2017-03-16): [AVR] Send USB ZLP if required
Fixes #5732 , "reverts" #4864

Tested with all the following problematic code on Linux, WinXp and Win10

https://github.com/NicoHood/HID/blob/master/examples/RawHID/RawHID/RawHID.ino (only on Linux)

```arduino
void setup() {
  // put your setup code here, to run once:

}

void loop() {
  if (Serial) {
    if (Serial.available() > 0) {
      Serial.read();
      char buff[65] = "D_ENDPOINT(USB_ENDPOINT_IN (CDC_ENDPOINT_IN ),USB_ENDPOINT_TYPE";
      Serial.write(buff,64);
      Serial.flush();
    }
  }
}
```

```arduino
void setup() {
  // put your setup code he [... cut]
closed 2017-03-16 by cmaglie