arduino/Arduino #5733: rollingAverage function in Average.h library does not work

marciukasb, 2016-12-16:
I have imported the Average library but none of the functions seem to work and even the examples in the package do not compile.

Chris--A, 2016-12-17:
>I have imported the Average library

From the  library manager?
If so you should find the GitHub repo for that library and post the issue there.

facchinm, 2016-12-19:
The Average library does not seem to be included in library manager. The closest one I could find is https://github.com/MajenkoLibraries/Average, which I tested and works well. If you could provide a link to the library we can try to understand if it's a core-related problem.

marciukasb, 2017-02-13:
I have tryed two different versions of arduino and still none of the methods work. The description in http://playground.arduino.cc/Main/Average says that the library https://github.com/MajenkoLibraries/Average has a method RollingAverage that adds values to the array shifting all the values down one place and returns the average. I cannot find or use this method.

per1234, 2017-07-01:
The new function for a rolling average in that [Average library](https://github.com/MajenkoLibraries/Average) is called [`rolling()`](https://github.com/MajenkoLibraries/Average/blob/master/Average.h#L113-L116), as was confirmed by the author here: https://github.com/MajenkoLibraries/Average/issues/8.

I have removed the obsolete documentation from http://playground.arduino.cc/Main/Average.

closed 2017-07-01 by per1234