arduino/Arduino #5734: [website] Broken datasheet links on product pages

per1234, 2016-12-16:
The datasheet links on all the following product pages are broken:

- https://www.arduino.cc/en/Main/ArduinoBoardUno
- https://www.arduino.cc/en/Main/Robot
- https://www.arduino.cc/en/Main/ArduinoBoardMegaADK
- https://www.arduino.cc/en/Main/ArduinoBoardPro
- https://www.arduino.cc/en/Main/ArduinoBoardEthernet
- https://www.arduino.cc/en/Main/ArduinoBoardLilyPad
- https://www.arduino.cc/en/Main/ArduinoBoardYun - The link is missing but the "ATmega32u4 (datasheet)" in the text indicates that one was intended.

Originally reported at http://forum.arduino.cc/index.php?topic=442344

akash73, 2016-12-19:
@per1234  thanks for notice.
we are managing it, I'll write once done.

thanks

akash73, 2016-12-22:
Hi, pages are now corrected.
thanks

closed 2016-12-23 by cmaglie