arduino/Arduino #5738: Arduino documentation needs correction to LOW (and also HIGH) voltage descriptions

ElectricRCAircraftGuy, 2016-12-19:
Documentation here (https://www.arduino.cc/en/Reference/Constants) states an input is HIGH if:

a voltage greater than 3 volts is present at the pin (5V boards);
a voltage greater than 2 volts is present at the pin (3.3V boards);

and LOW if:

a voltage less than 3 volts is present at the pin (5V boards);
a voltage less than 2 volts is present at the pin (3.3V boards);

This doesn't look quite right to me. The most recent ATmega328 datasheet from 11/2016 (http://www.atmel.com/Images/Atmel-42735-8-bit-AVR-Microcontroller-ATmega328-328P_Datasheet.pdf) says in Table 32-2 (pg. 365) that V_IH is 0.6Vcc for Vcc = 2.4 to 5.5V, and that V_IL is 0.3Vcc for Vcc = 2.4 to 5.5V. 

Can we please update the documentation to update this? Or am I wrong?

I recommend we make the documentation say:

------------start of documentation-------------

An input is read as HIGH if the voltage is > 0.6 x Vcc, and LOW if the voltage is < 0.3 x Vcc, for Vcc voltage levels ranging from 2.4V to 5.5V. Reference the ATmega328 datasheet, revision 11/2016, Table 32-2, pg. 365.

That equates to the following:

HIGH if
-a voltage greater than 0.6 x 5 = 3.0V is present at the pin (5V boards);
-a voltage greater than 0.6 x 3.3 = 1.98V is present at the pin (3.3V boards);

and LOW if:

-a voltage less than 0.3 x 5 = 1.5V is present at the pin (5V boards);
-a voltage less than 0.3 x 3.3 = 0.99V is present at the pin (3.3V boards);

------------end of documentation--------------

septillion-git, 2017-01-19:
But then you would need to explain the gap of 0,3 x Vcc as well.

Also I would round the number to 1 decimal. Aka 3.0V, 2,0V 1,5V and 1,0V. 0.99V is just confusing.

akash73, 2017-01-19:
@ElectricRCAircraftGuy 
Thanks for pointed out .

Page updated

closed 2017-01-19 by akash73

septillion-git, 2017-01-19:
@akash73, can you please update the 1.98V to just 2.0V. There are just not enough significant figures to justify 1.98. And besides, an exact figure like that can be confusing for newbies.

And even op top of that, for a SAM board it's 0,7 x Vcc to be a HIGH...

So I think the article should or be simplified and just say High is close to Vcc and low is close to 0. Or it should be mentioned this is for the standard Arduinos and it may differ on the more complex boards.