arduino/Arduino #5742: Connection Error

najkulime, 2016-12-19:
Unable to contact the code compilation service. 

Please try again later or contact us if the problem persists.

I have tryed again later but this still appears. Any suggersions? It refers to this project:

https://circuits.io/circuits/3557330-the-unnamed-circuit

matthijskooijman, 2016-12-19:
What is this about exactly? If it's about circuits.io, that's not related to Arduino at all. Or is this about the online Arduino IDE? Can you be more specific? Perhaps a screenshot of the error?

najkulime, 2016-12-20:
[image]

facchinm, 2016-12-20:
Hi @najkulime ,
indeed you should contact circuits.io guys (maybe via their support). we are not the right people to solve your problem :smile:

closed 2016-12-20 by facchinm