arduino/Arduino #5750: menu font looks like broken in ide-1.8.0

Myriads, 2016-12-23:
menu font looks like broken in ide-1.8.0:
[image]
[image]

facchinm, 2016-12-23:
Hi @Myriads ,
can you expand your bug report? Are you referring to Korean vs English for board names and programmers or something more subtle?

Myriads, 2016-12-23:
[image]

[image]

SurfingDude, 2016-12-23:
Same problem in the English Language IDE. Going from 1.6.12 to 1.8.0 gives thinner letters in a font that is more difficult to read. The preferences dialog shows this quite clearly.

cmaglie, 2016-12-23:
Which version of windows are you running?

SurfingDude, 2016-12-23:
I am running Windows 10 Pro.

      ...glen

From: Cristian Maglie 
Sent: Friday, December 23, 2016 9:03 AM
To: arduino/Arduino 
Cc: Glen Johnson ; Comment 
Subject: Re: [arduino/Arduino] menu font looks like broken in ide-1.8.0 (#5750)

Which version of windows are you running?

cmaglie, 2016-12-23:
For some reason the antialias is gone.
Investigating...

cmaglie, 2016-12-23:
It seems caused by this commit cf54cce that is part of #5578.
I'm looking for a solution.

DougieLawson, 2016-12-23:
@cmaglie Do you need any additional docs, screenshots or stuff from a Windows 10 user?

Do you need anyone to alpha or beta test any fix you develop for this? Give me a download link and I'm quite happy to give things a try out (although not on Xmas day).

cmaglie, 2016-12-23:
I managed to reproduce it on my pc so I'm ok for now, thanks!
A test build to try will be produced by @ArduinoBot once I have a fix.

FernandoGarcia, 2016-12-24:
Hi @cmaglie !

This PR solve the problem to me.

Thanks.

Best regards.

closed 2016-12-28 by cmaglie

ameliaamelia, 2016-12-28:
I am using Arduino IDE 1.8.0 and Visual Studio 14 on Windows 10.
The font for both is "Input"; with IDE 1.6.13 all OK but with IDE 1.8.0 the fixed length is not taken in account:
SHORT  a;
BYTEu  b; (note the 'b' is slightly left shifted, like in this non fixed font post)
The font is correctly displayed.

amelia

ameliaamelia, 2016-12-28:
Addendum:
If I use the font "Monospac821 BT" all is OK.

amelia

DBenS, 2016-12-31:
Hi,
I'm using Windows 8.1 and Arduino IDE for Windows.
The editor has its font OK, but the message window is really very bad. I tried to change the font settings onto "preferences.txt" without any effect.

See below:
[image]

PaulStoffregen, 2016-12-31:
@DBenS - This problem was fixed 3 days ago.  Download the latest nightly build version to get the fix.

PR #5756 (2016-12-28): Fix font anti-alias on windows
The properties:

      System.setProperty("awt.useSystemAAFontSettings", "on");
      System.setProperty("swing.aatext", "true");

actually works on Linux (where the hint helps X11 to enable antialiased rendering) but makes things worse on Windows where the outcome is exactly the opposite (anti-alias is disabled).

Previously those settings had no effect because they were executed *after* the initialization of the graphics. This is no more true after the merge of #5578, that moved the graphics initialization after commmand line parsing and consequently revealed t [... cut]
commit 34fb8027 (2016-12-28): Fix font anti-alias on windows

The properties:

      System.setProperty("awt.useSystemAAFontSettings", "on");
      System.setProperty("swing.aatext", "true");

actually works on Linux (where the hint helps X11 to enable antialiased
rendering) but makes things worse on Windows where the outcome is exactly
the opposite (anti-alias is disabled).

Previously those settings had no effect because they were executed
*after* the initialization of the graphics. This is no more true
after the merge of #5578, that moved the graphics initialization
after commmand line parsing and consequently revealed  [... cut]