arduino/Arduino #5755: SDClass::begin broken between 1.66 and 1.80

Geezerjem, 2016-12-23:
SDClass::begin now has two entries in SD.cpp .

boolean SDClass::begin(uint8_t csPin) {
  /*
    Performs the initialisation required by the sdfatlib library.
    Return true if initialization succeeds, false otherwise.
   */
  return card.init(SPI_HALF_SPEED, csPin) &&
         volume.init(card) &&
         root.openRoot(volume);
}
boolean SDClass::begin(uint32_t clock, uint8_t csPin) {
  return card.init(SPI_HALF_SPEED, csPin) &&
         card.setSpiClock(clock) &&
         volume.init(card) &&
         root.openRoot(volume);
}

It doesn't work. The first entry was there in my v1.66 and it worked fine. Two entries have appeared in v1.80 and a initializing the sd card with an if (!SD.begin(SD_CS)) fails every time.

Jem.

sandeepmistry, 2016-12-23:
@Geezerjem what board and hardware are you using?

Geezerjem, 2016-12-24:
@sandeepmistry I'm using an Esplora and an 'Arduino compatible' LCD module. Coming back to it this morning, it fails to compile,
Arduino: 1.8.0 (Windows 7), Board: "Arduino Esplora"

C:\Program Files\Arduino\libraries\SD\src\SD.cpp:350:9: error: prototype for 'boolean SDLib::SDClass::begin(uint32_t, uint8_t)' does not match any in class 'SDLib::SDClass'
 boolean SDClass::begin(uint32_t clock, uint8_t csPin) {
         ^
C:\Program Files\Arduino\libraries\SD\src\SD.cpp:337:9: error: candidate is: boolean SDLib::SDClass::begin(uint8_t)
 boolean SDClass::begin(uint8_t csPin) {
         ^
Using library Esplora at version 1.0.4 in folder: C:\Program Files\Arduino\libraries\Esplora 
Using library SPI at version 1.0 in folder: C:\Program Files\Arduino\hardware\arduino\avr\libraries\SPI 
Using library SD at version 1.0.6 in folder: C:\Program Files\Arduino\libraries\SD 
Using library TFT at version 1.0.6 in folder: C:\Program Files\Arduino\libraries\TFT 
exit status 1
Error compiling for board Arduino Esplora.
I understand the Esplora is unsupported, but looking at the current SD examples, and compiling for my UNO, they all have a single argument for SD.begin and also fail to compile.

q2dg, 2016-12-25:
Well... in fact... why not https://github.com/arduino-libraries/SD/issues/6 ?

sandeepmistry, 2016-12-29:
Hi @Geezerjem,

I've just tried build all the [SD (v1.1.0) library examples](https://github.com/arduino-libraries/SD/tree/master/examples) with the Explora board selected on my Mac using IDE 1.8.0. No compile errors occurred.

Could you please provide a sketch that is causing the build error. It also looks like the IDE is selecting an older SD version `Using library SD at version 1.0.6`.

sandeepmistry, 2017-01-11:
> the copy append on my case when I've roll back version because the 1.1.0 version fail to initialize with begin(ssPin)... (don't know if it is related...)

@NitrofMtl there's a new 1.1.1 release with a fix for this.

facchinm, 2017-01-12:
You can update SD library standalone via Board Manager or update to IDE 1.8.1 (which bundles the new library and solves the "ugly fonts" problem on Windows)

closed 2017-01-12 by facchinm