arduino/Arduino #5757: L10N for compilation message - IDE1.8.0

Myriads, 2016-12-24:
The previous version provides the L10N for compilation message but IDE1.8.0 doesn't provide:

> Sketch  uses 2660 bytes (8%) of program storage space. Maximum is 32256 bytes.
Global variables use 208 bytes (10%) of dynamic memory, leaving 1840 bytes for local variables. Maximum is 2048 bytes.

facchinm, 2017-01-02:
Hi  @Myriads, that shouldn't happen since we preserved the wording when moving the sketch size calculation from the IDE to `arduino-builder`.
If I set the IDE language in Korean I get
```
스케치는 프로그램 저장 공간 9056 바이트(3%)를 사용. 최대 262144 바이트.
```
Is it correct? How can I replicate your configuration?

Myriads, 2017-01-03:
Hi @facchinm. The message you got is correct. But in my case, evenif I set the IDE language to Korean, I always get the English messages.

Myriads, 2017-01-03:
I replicated this. Please build resource files with update.sh and rebuild IDE, then messages are displayed in English.

facchinm, 2017-01-03:
You are totally right; indeed 1a97ec448182c57bc8fc0278075f429f3877b2a2 removed the Java strings but the actual resources in 1.8.0 are correct (still don't know why). Anyway, I'm pushing the strings back and recreating the resources. Thanks for helping!

PR #5803 (2017-01-04): Fix translatable strings removed before 1.8.0
Strings were removed by https://github.com/arduino/Arduino/commit/1a97ec448182c57bc8fc0278075f429f3877b2a2 and https://github.com/arduino/Arduino/pull/5578
Fixes #5757
commit 35ee2499 (2017-01-04): Readd translatable strings

Fixes #5757
closed 2017-01-04 by cmaglie