arduino/Arduino #5758: Arduino.cc SSL Certificate expired

dlabun, 2016-12-24:
I think that pretty much sums up the issue, the SSL cert expired today.

NicoHood, 2016-12-24:
It seems that this only applies to downloads.arduino.cc. However I cannot build arduino packages anymore and I will not download them via http as no GPG signatures #5619 are available.

Test URL:
https://downloads.arduino.cc/cores

dlabun, 2016-12-24:
You're right, I blew right through the homepage and didn't even realize it was a HTTPS connection. Somebody must have forgotten to update all of the subdomains with the new cert.

Chris--A, 2016-12-24:
I'm getting the issue on forum.arduino.cc. It seems now it is forced over a https connection. Can't receive any data via http.

PaoloP74, 2016-12-24:
Dear Santa Claus, please bring a new certificate to the Arduino Team.

PaulStoffregen, 2016-12-24:
The forum also seem to be affected.

Isaac100, 2016-12-24:
Yep, it definitely expired.
[image]

Isaac100, 2016-12-24:
I've figured out a temporary fix, set your computer's clock to before the expiration date(Dec 24).  On Firefox it should go automatically to the website and on Chrome the same error will come up, just click on advanced and click continue to website.

matteosuppo, 2016-12-24:
Santa arrived with a brand new certificate

mastrolinux, 2016-12-24:
Solved, thanks for the report

closed 2016-12-24 by mastrolinux