arduino/Arduino #5762: Driver signing certificates expired

dlabun, 2016-12-26:
While installing IDE 1.8.0 on Windows I found that the signing certificates for the Linino boards have been expired for over a year. All of the Arduino drivers will be expiring over the next weeks.

Coding-Badly, 2016-12-26:
What was the error message?
When was the error message displayed?

dlabun, 2016-12-26:
Who said anything about an error message?

Coding-Badly, 2016-12-26:
Then why are you complaining the signing certificates are expired / expiring?

dlabun, 2016-12-26:
There was no complaining but rather a factual issue reported to Arduino.

Coding-Badly, 2016-12-26:
Ugh.  Stop playing childish games.

Why do you believe an expired / expiring signing certificate is an issue?

PaulStoffregen, 2016-12-26:
Indeed, this isn't anything like the recently expired SSL certificate on the website, where users will be impacted.  With driver signing, the certificate expiration merely means new drivers can't be signed.  Whoever does the signing may have already renewed their certificate, for all we know.  If they haven't, they'll do so when they need to sign drivers again.

dlabun, 2016-12-26:
@Coding-Badly If you want to be unprofessional, that's fine with me as I'll just let Github support deal with you. BTW, for someone that went on a rant about leaving Github over harassment it's funny you're being childish over an issue report for a project you're not associated with.

@PaulStoffregen 1.8.0 will start throwing driver signature errors if you opt to have the installer install the drivers (which I normally don't). It appears the linino-boards_x86, slabvcp and cdc_vista drivers were all countersigned by now revoked certificates. These drivers did not appear in 1.6.12.

Coding-Badly, 2016-12-27:
> 1.8.0 will start throwing driver signature errors...

Only if the signed file does not include a timestamp.  Which has nothing to do with when the signing certificate expires.

You have yet to provide any evidence that the expiration is an actual issue.

NicoHood, 2016-12-27:
@Coding-Badly man chill! He wanted to let the arduino team know the issue. So if any problem MAY occur next time we do not have thousand of failed installations on windows (similar to the website https issue). It was a hint, no matter if the world end with arduino tomorrow or not. Good that he let us know.

cmaglie, 2016-12-27:
The drivers are timestamped, so the signature will still be valid even if the certificate used for signing expires, because the timestamp guarantee that the signature has been done before the certificate expiration.
The worst that can happen is that we can't sign *new* drivers, but this will produce an error early in the codesign tool during the build.

This is different from the SSL certificates issue where the update procedure failed to propagate the renewed SSL certificate to the arduino.cc subdomains. In that case you (or a cron job) must *actively* replace expired certificates.

Thanks for the report BTW!

PaulStoffregen, 2016-12-27:
Just for fun, I re-imaged my Windows test machine to a clean Windows 7, and then ran the Arduino 1.8.0 installer.  Looks like @dlabun has a point.  Seven driver install dialogs appeared, some of them the unsigned driver warning.

PaulStoffregen, 2016-12-27:
Here are screenshots of the 7 dialog boxes, in their order or appearance, I saw while installing to a clean system.

[image]

[image]

[image]

[image]

[image]

[image]

[image]

It's not my intention to "stir the pot" in this already somewhat unpleasant conversation.  I'm going to refrain from commenting on *why* this happens.

But hopefully these first-time user experience screenshots help?

cmaglie, 2016-12-27:
Oh! thanks I'm doing about the same thing (except that my VM is currently stuck doing updates...).
So it looks like @diabun was right after all :-). I'll check what's happening with the drivers that fail signature verification.

Coding-Badly, 2016-12-30:
dpinst-amd64.exe needs to be re-signed using SHA 256.

cmaglie, 2017-01-05:
Ok it seems that some drivers, in a way or another, have had a bad signature:

CP210x -> some files missing and not up-to-date
https://github.com/arduino/Arduino/commit/2daba3bb9337533d956fe0d7b3b19a689df47c58
https://github.com/arduino/Arduino/commit/694dc96159b0d18b563c4af2c6846380a930fafb

Adafruit playground -> signature in sha1 but timestamped in sha256, it seems that both signature and timestamp needs to be sha1
https://github.com/arduino/Arduino/commit/123c231c8e8c9b1d610d4ad26a294830c624e1bc

Linino One -> fixed in #5809, don't know the exact details (updated also arduino-org drivers in the same PR)

> dpinst-amd64.exe needs to be re-signed and re-timestamped using SHA 256

this file has been signed by Microsoft, just used as-is.

closed 2017-01-05 by cmaglie

per1234, 2017-02-11:
Maybe this should be an issue at https://github.com/arduino/ArduinoCore-samd instead but it seems related to this thread. I just updated Arduino SAMD Boards to 1.6.12 and got this:
[image]

dlabun, 2017-02-11:
I already reported it a few weeks ago on the SAMD thread and it's closed as fixed in release 1.6.12 for the SAMD...

https://github.com/arduino/ArduinoCore-samd/issues/206

per1234, 2017-02-11:
Ah, I had seen that issue but didn't realize the Adafruit board was part of the Arduino SAMD Boards package, I though that was referring to Adafruit's hardware package. It does seem that the issue was not fixed.