arduino/Arduino #5764: Probelm with MKR1000 Documentation (Blink Example)

MaxiBerdy, 2016-12-26:
I was trying to make the integrated led work on my MKR1000 with the Blink example and with a little of search on Google, I fount out that the led pin is not pin 13, but pin 6 !! You should mention it somewhere on the doc, because all is written is to use the exemple code and start it. Thanks again !

q2dg, 2016-12-27:
It's strange because here is well documented: https://github.com/arduino/Arduino/blob/08af7f8991e312feeaea01fb3943eb4e28ac6a71/build/shared/examples/01.Basics/Blink/Blink.ino

oqibidipo, 2016-12-27:
> well documented

... starting from IDE version 1.6.12.

cmaglie, 2016-12-27:
@MaxiBerdy 
the Blink example has been updated to reflect this, and now it uses the `LED_BUILTIN` costant, BTW this is true from the IDE >=1.6.12. Moreover the documentations states:

*Onboard LED
On MKR1000 the onboard LED is connected to D6 and not D13 as on the other boards. Blink example needs to be changed to work properly.*

https://www.arduino.cc/en/Main/ArduinoMKR1000

Maybe you have an old version of the IDE installed? BTW if you think that we should write that more clearly feel free to suggest how/where.

per1234, 2016-12-27:
> Blink example needs to be changed to work properly.

It seems like that page should be updated now that the Blink sketch has been updated a couple of IDE releases ago.

cmaglie, 2016-12-27:
Yes, but it's problably best to leave the warning, so users can double check that the example is correct. Something like:

*Blink example **or other sketcthes that uses pin 13 for onboard LED may** needs to be changed to work properly*

per1234, 2016-12-27:
I agree. I just encountered someone on the forum who was confused whether they should be using pin 13 for the LED on their MKR1000. Maybe something like:
>The Blink example(Arduino IDE <= 1.6.11) or other sketches that use pin 13 for onboard LED may need to be changed to work properly

MaxiBerdy, 2017-01-08:
https://www.arduino.cc/en/Guide/MKR1000
On this page, there are no information about the pin to use on the MKR1000 ! Thats why I didn't understand..

cmaglie, 2017-01-08:
oh I see, thanks for pointing this out.

/cc @akash73

SurfingDude, 2017-01-08:
The BlinkWithoutDelay example still hard codes:
const int ledPin =  13;      // the number of the LED pin
Also example DigitalInputPullup hardcodes  pin 13

akash73, 2017-01-09:
thanks for pointing out.

@MaxiBerdy we've added a note on the Getting started page.
@SurfingDude thanks for notice, we will update the example soon.

closed 2017-01-09 by cmaglie