arduino/Arduino #5773: No support for silent install/uninstall of Windows-application (not drivers)

emipa606, 2016-12-28:
Hi

The installer for Windows throws dialog boxes when in silent mode.
`arduino-windows.exe /S`
[image]

When trying to uninstall silent as well.
`"%programfiles(x86)%\Arduino\uninstall.exe" /S`
[image]

These custom dialogs makes it impossible to update the software silently. 
For examples on how to make the installer silent there is a good post on the issue [here](http://stackoverflow.com/questions/14445395/nullsoft-scriptable-install-system-nsis-installer-silent-mode)

_NOTE: This is not an issue with the driver-installations. I have already solved that by importing the driver-certificates before installation using:_ 
`certutil -addstore "TrustedPublisher" "%cd%\Arduino-LLC.p7b"`
`certutil -addstore "TrustedPublisher" "%cd%\Arduino-SRL.p7b"`

facchinm, 2017-01-02:
A possible solution is to patch the `nsis` script we are using to be aware of /S flag (http://nsis.sourceforge.net/Docs/Chapter4.html#ifsilent). @cmaglie, is "silent" uninstallation of previous version acceptable?

facchinm, 2017-01-02:
@emipa606 could you test https://downloads.arduino.cc/ide_staging/arduino-1.8.0-windows-silent.exe and report if the behaviour is correct? Thanks!

emipa606, 2017-01-02:
Tried it and when I ran
`arduino-1.8.0-windows-silent.exe /S`
I got this:
[image]

So it seems the first warning is gone but I assume the installer calls the uninstaller for the previous version witch still shows the message. Is there a way to bypass this?

facchinm, 2017-01-02:
Correct, the first time the old uninstaller gets called, and I'm not aware of any workaround... Any subsequent call works as expected but probably it's not what you need :slightly_smiling_face:

emipa606, 2017-01-02:
Since I have this installed on multiple computers (I am a systems administrator at a university) i was hoping that there was a way for me to replace the earlier version without manual work on each installation.
Perhaps I can remove a registry value/file to make the installation unaware of the earlier version?

Can I assume that the new version will be silent aware in the future after this?

facchinm, 2017-01-03:
Yep, it's currently in staging but if it works I'd merge it. If it can help, the line used by nsis to discover if the IDE is already installed if 
```nsis
    ; Check again for an already-installed Arduino.  This will catch the case where more than
    ; one version is installed and only one was removed as well as the case where the user
    ; or system aborted the uninstaller before it could complete.
    ReadRegStr $R0 HKLM "Software\Microsoft\Windows\CurrentVersion\Uninstall\Arduino" "UninstallString"
```

Closing as fixed, reopen if needed

closed 2017-01-03 by facchinm

emipa606, 2017-01-03:
Thanks!

Got a working install-script now if anyone wants to use it for deployments:
`reg delete "HKLM\Software\Microsoft\Windows\CurrentVersion\Uninstall\Arduino" /f`
`certutil -addstore "TrustedPublisher" Arduino-LLC.p7b`
`arduino-windows.exe /S`