arduino/Arduino #5780: The program 'arduino' is currently not installed

franciscop, 2016-12-29:
After following a similar guide to* [the official guide](https://www.arduino.cc/en/Guide/Linux) I try to execute the [CLI explained here](https://github.com/arduino/Arduino/blob/master/build/shared/manpage.adoc) but I get the following error:

> The program 'arduino' is currently not installed. You can install it by typing:
> sudo apt install arduino

Which is probably highly likely to introduce the user into error as then the user will try to install the highly outdated Arduino version. I know this is the wrong arduino so I didn't do that, but it should probably be explained somewhere for the unitialized user or even included into the tutorial.

To add it to the user path, I executed this script to create a symbolic link to the `bin` folder. I'm not sure whether this is the best way or not, but it does work as expected:

```bash
$ sudo ln -s /opt/arduino-1.8.0/arduino /usr/bin/arduino
```

This should most likely be included into the `./install.sh` script with some logic for the version number instead of requiring the user to take an extra step, that's why I'm opening this issue.

* The only difference was that I moved it first to `/opts`, a place probably better suited for the installation files than `~/Downloads`

NicoHood, 2016-12-29:
This is a problem of your distribution and not arduino. Please contact your distribution packagers to fix arduino for you. The installation script is something you really do not want to use unless you want to trash your system. ArchLinux gives you examples on how to package arduino.

matthijskooijman, 2016-12-29:
The install script only takes care of adding a menu shortcut, so Arduino should be listed in your desktop environment's menu. To allow starting arduino on the commandline, you need to specify a full path to it, or make sure that the arduino executable is in one of the directories in the `$PATH` environment variables (which you can fix by either changing `$PATH`, or creating a symlink to the arduino executable in a directory that is already in `$PATH`, such as `/usr/local/bin`).

I think you should have sufficient info to figure this out from here, so I'm closing this ticket. If you have further questions or suggestions on documentation improvements, feel free to leave more comments here.

franciscop, 2016-12-30:
@matthijskooijman please read the original issue; I solved this without problem as I know my way around Linux (in the same way you explained). However, **the official guide is incomplete** and that's why I opened this issue.

@NicoHood *The installation script is something you really do not want to use unless you want to trash your system* => but **the installation script is the official installation method** according to Arduino's guide/documentation.

matthijskooijman, 2016-12-30:
@franciscop, apologies, I should have read your issue more closely.

Would you perhaps like to propose some text to be added to the guide to cover this? I think that it should both explain what the install script does exactly (and point out that it does not actually copy the Arduino files anywhere, so the unpacked files should stay in place) and point out how to put things into `$PATH` (adding a symlink as you did is probably the best option for that, except it's better placed in `/usr/local/bin` I think).

Alternatively, we could change the install script to create this symlink. For that, it would probably need to check the `$PATH` value to see if `/usr/local/bin` is in there, and then create a symlink. Is  that something you would want to provide a pull request for?

> The installation script is something you really do not want to use unless you want to trash your system.
@NicoHood I'm not sure what you mean by that. Unlike most install scripts, this doesn't actually copy all the files into /usr or wherever, so it's quite a non-invasive script already

franciscop, 2016-12-30:
@matthijskooijman I don't feel capable of editing that script for production (if I screw my system that's okay, but I don't want to break other's machines). I could add in the documentation what I did though, and edit what is already in there to reflect a better directory than `~/Downloads` to keep the installation files.

Could you please point how to edit [Install the Arduino Software (IDE) on on Linux](https://www.arduino.cc/en/Guide/Linux)? I just registered expecting to see an "Edit" button but couldn't find any and searching this repository didn't bring up much.

per1234, 2017-02-01:
>Could you please point how to edit Install the Arduino Software (IDE) on on Linux?

Just post your suggested changes here and it will be assigned to someone who has the ability to edit the website.

facchinm, 2019-03-07:
Fixed by #7593

closed 2019-03-07 by facchinm